df8oe/UHSDR #1109: Stuttering Tuning on USB/LSB after software update

g7jiq, 2017-10-04:
After updating to the latest software D2.5.117.
When I tune USB or LSB the frequency readout momentary flashes red with a clunk in the sound.
This is worse the faster you tune or bigger steps taken.
AM is unaffected and this happens on any band.
Pria to the update all was fine, I have rolled back to a earlyer ver and the problem still exists.

Mayby the update upset something in the menu.
NB-: I am running the IC2bus 1 and 2 at 100khz as before.

Has anyone any ideas ???

DG9BFC, 2017-10-04:
hmm ... i can select whatever filter ... and tune up and down ... and all is normal here
even very fast tune with dynamic tune switched on ... and i have no problem
i2c is running at 320 here (yes i know that is fast setting with red numbers ... but it works fine)

df8oe, 2017-10-04:
Maybe your config is corrupted by chance. Do you have backup in virtual EEPROM?

df8oe, 2017-10-05:
Any progress?

DG9BFC, 2017-10-05:
he should do a hard reset of the config (start with f1 f3 f5 pressed)

g7jiq, 2017-10-05:
Will do a reset and let you know what happens.

DG9BFC, 2017-10-12:
any progress??

closed 2017-10-21 by g7jiq