df8oe/UHSDR #1115: TX frequency calibration

degaus72, 2017-10-05:
Hello,

mcHF v0.5 - bootloader 3.4.1 - firmware 2.5.118

While I was calibrating the TX frequency, I have noticed some discrepances between bands.
I have started to adjust the frequency on 80m following the manual and that was ok with a ppm of +10,4, but when I test on 40m freq. 7.184.000 the counter shows 7.183.249, then if I move to 20m freq. 14.244.000 it shows 14.243.997.

The 3Hz difference on 20m maybe be acceptable but the the 751Hz on 40m probabily not. I did try to adjust the 20m band but it messed up the 80m, am I missing some steps?

73 de M0TKM

db4ple, 2017-10-05:
Hi,

I am sure you had CW on 40m, this causes your perceived deviation. Use the same mode on all bands gives consistent frequency measurements.

73
Danilo

degaus72, 2017-10-05:
Hi Danilo,
I have the same settings in all the 3 bands.
I have attached some pictures.
73, Matteo

db4ple, 2017-10-05:
You should respond through the Webinterface, I can't see the pictures. And when responding via mail, please remove the text you respond to (otherwise the comments will be cluttered with comment quotes).
TU
Danilo

degaus72, 2017-10-05:
Hi Danilo,

Sorry about that. First time I use the github webinterface.
Pictures attached....hope :-)

73, Matteo

[image]
[image]
[image]
[image]
[image]
[image]

df8oe, 2017-10-05:
You are transmitting on 40m in LSB. TUNE uses sidetone: voilá - there is your offset!

You must use FM on ALL bands.

degaus72, 2017-10-05:
What an i****t I am!! I did check the settings at least 4/5 times before asking for help!!
Sorry for wasting your time...all ok now just 2/3 Hz difference.

73, Matteo

db4ple, 2017-10-05:
Hi Matteo, nothing more to comment.

closed 2017-10-05 by df8oe