df8oe/UHSDR #1119: Tuner Helper

WD8BXS, 2017-10-10:
I have loaded Firmware D2.5.116

I do not see a yellow bar in the tuner helper window.
Am I missing something

Thanks,
 Chuck WD8BXS

df8oe, 2017-10-10:
Do you try to tune to any signal? Marker is first drawn if any signal is received. Then it stays forever (even without signal).

vy 73
Andreas

WD8BXS, 2017-10-13:
I have 2 radios here and neither one has the yellow bar in the tuner helper!!

df8oe, 2017-10-14:
1) Switch to CW
2) activate "Tune helper" in configuration menu
--> you will get a small green box above DSP box - usually EMPTY.
3) Tune to any CW signal. It is neccessary to set sidetone frequency to an reasonable value !=0.
--> The nearer you get to the sidetone frequency while tuning you see a yellow marker which stays in the middle of the box if your received audio is exactly sidetone frequency.

WD8BXS, 2017-10-14:
Yes Andreas, 

Here is a video of the radio on CW, tone is 750, tuner helper on.

I have the green box, and it is always empty.
As you see in the video , it never changes as I tune through the station, you will also see that it the CW is copied just fine.

https://1drv.ms/v/s!AhAcG-wkckUviW6NOqmtbBGcDVYm

Chuck

From: DF8OE 
Sent: Saturday, October 14, 2017 3:39 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Tuner Helper (#1119)

  1.. Switch to CW 
  2.. activate "Tune helper" in configuration menu
  --> you will get a small green box above DSP box - usually EMPTY. 
  3.. Tune to any CW signal. It is neccessary to set sidetone frequency to an reasonable value !=0.
  --> The nearer you get to the sidetone frequency while tuning you see a yellow marker which stays in the middle of the box if your received audio is exactly sidetone frequency.

df8oe, 2017-10-14:
I am nearly sure you do have a setting otherwise than most of the users: We all do have in display menu "dBm display": to dBm - you do have it OFF - isn't it?

73
Andreas

WD8BXS, 2017-10-14:
AH, that was it my Friend.

Thank you so much!!

Chuck

From: DF8OE 
Sent: Saturday, October 14, 2017 10:45 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Tuner Helper (#1119)

I am nearly sure you do have a setting otherwise than most of the users: We all do have in display menu "dBm display": to dBm - you do have it OFF - isn't it?

73
Andreas

closed 2017-10-20 by WD8BXS