df8oe/UHSDR #1124: Local <-Remote update of Clone

Airheadbit, 2017-10-12:
I've been using GitKraken but I noticed my local (files on my local hard drive) were not the same. Everything I've found in GitKraken says Fetch get updates but doesn't update my 'Local' hard drive? Pull fast-forward?

Is there a command line git statement that would update my clone and NOT screw with the 'Remote'?
Thanks
Kevin K2AAE

db4ple, 2017-10-12:
No.  This is not how Git works. 

It is always
Remote Git Repo -> 'fetch' -> Copy of Remote Git Repos  in Local Repo -> 'Merge/Rebase' -> changes to a Branch in Local Repo.  

Pull is basically a combined fetch then merge.

I don't know GitKraken, to be honest. But if they follow the official git wording, this it how it works.

K2AAE, 2017-10-12:
Contribution.md has broken links.
The issue was I cloned dof8e/UHSDR to my local HD. I have to remove it and clone K2AAE/UHSDR which is a FORK (dof8e/UHSDR top right Fork into my github account, about three seconds) of dof8e/UHSDR.
So to start the process, fork dof8e/UHSDR to a github account then clone that new git to your local hard drive to work with... and to think I spent two hours surfing the web trying for find how to do something that can't be done since the initial setup was forked...

closed 2017-10-12 by K2AAE