df8oe/UHSDR #1125: Automated Builds -

Airheadbit, 2017-10-12:
We've talked about this. I put up Forum and added some radio check boxes that captures what the user wants, emails/ftp/curls/TCP/UDP it to a server, the server syncs the git, builds the requested firmware, ftp's the file back and emails the user with a link to the firmware. The Forum can weed out spammers, is multi-language (English and German - what else?) and we can have a place to discuss different topics, using this git:issues isn't it. I secured a short name for a domain osdr.us - any suggestion or objections?
Need Forum Section suggestions: Display, FreeDV, CW, etc....
Thanks
Kevin K2AAE

df8oe, 2017-10-12:
Hi Kevin,

before we could/should start this we should state and discuss what we want to reach. Forum is already existing and it does not make sense to scatter something. About builds: I already thought about "automated builds". This can be independendent to my server. Help is appreciated. But Links should be related to github. So if you want to setup a compiler, building a radio-button web interface for building the binaries, fetching automatically the newest sources from GitHub: let us discuss the path we want to go.

I am open to everyone who wants to contribute and contribution (whatever kind) is welcome!

73
Andreas

K2AAE, 2017-10-12:
Where is the forum? Discussion and news board of the DARC local branch I40?
You're going to use the radio clubs forum for all version of this project?

K2AAE, 2017-10-12:
I have a Dual Xeon server sitting here that is no longer in use, figured to load Linux and let it cron jobs. 
"before we could/should start this we should state and discuss " that was the reason for this thread.

phaethon, 2017-10-13:
JetBrain offers enterprise version of TeamCity for free to open source projects. I am currently considering it for another project. CLion (which I like much more than eclipse), too, but it uses CMake and would require rewriting make scripts.

df8oe, 2017-10-13:
@K2AAE : Yes - we will use web pages of OVI40 for OVI40 project and for existing mcHF project. First mcHF building started in 2015 and supported many German HAMs in building Chris' rig. OVI40 then was born as an idea to implement many new featurs and design a new radio. This will also be discussed at our forum. It is the forum of DARC OV I 40 - not of DARC himself. DARC main pages does not offer the needed functionalities. Our OV starts the project and will populate it to the world. BUT: We are working on a specific hardware we have choosen. It is absolutely destructive to think about implementation of new hardware before the given one is fully working. You will never get a working and usable rig if you always implement everything you have found somewhere on the web. So we will NOT discuss about using nucleo board - because this is not part of OVI40. And at this stage there is no need about new audio codecs - because the used ones do their job very well. I can extend the list much more. We have already the first statements on our web pages what OVI40 is intended for and what it will contain and provide.

If you want to start integration of nucleo board as "ui hardware" in UHSDR: you are welcome and invited to do so! We will merge your code if it does not hurt our main goals and so we add a "new supported hardware platform". Builds for mcHF and OVI40 must not be harmed, addition of new code which gives benefits for this two will be offered to all worlds and code which only works on specific hardware must be included in # ifdefs. And addition of code must follow our contribution rules. I am sure they are not complete and may be subject of modifications in the future - but we can and will discuss that by time.

K2AAE, 2017-10-13:
While I guess you will be removing the code for the 800X480 controller since there is no 3.5" displays that will fit the front panel of mcHF or OVI0 that can do 800X480 and new hardware is "absolutely destructive" to your project.

closed 2017-10-13 by K2AAE

df8oe, 2017-10-14:
Hi Kevin,

actually UHSDR supports two hardware platforms: mcHF and OVI40. If there is a simple way to connect the LCD you want to choos to the serial or SPI bus this "new hardware" is welcome. If it only runs using nucleo shield first there must be created a "new platform".

Regarding higher resolutions:
We are recently working on LCDs which do have a higher resolution than 320x240. For supporting different resolutions we will start a table-based GUI defintion where output positions and touchscreen positions will be defined for all supported resolutions. But we are not yet at this point: we are at the addition of new drivers...

UHSDR does not want to be highly experimental firmware for very special hardware combinations. It wants to be a firmware which supports hardware which can easily be reproduced (homebrew) or can be bought for "HAM prices". So we have set up our contribution rules (and they will extend if we are reaching a new problematic part). New implementation of hardware (or functions) must be compatible with existing hardware and functions. If they cannot be supported by all hardware platforms, code must be excludable by # ifdefs. Additions are not allowed to harm existing things! To be specific for your question:

How do you want to connect LCD to STM32?

Additional:
If you do want to play around with highly experimental hardware: you can do that - it is Open Source GPLv3. If you want to contribute the code later: rebase as often as you can with the active-devel so that it can be merged at a later time.

K2AAE, 2017-10-18:
"If you do want to play around with highly experimental hardware: you can do that - it is Open Source GPLv3. If you want to contribute the code later: rebase as often as you can with the active-devel so that it can be merged at a later time." The Rbase thing isn't working, still trying to figure this one out.

The generation of another hardware platform added 4,179 files, 83.5 MB to the git. 
GitKraken broke right after...

df8oe, 2017-10-19:
Rebase is working perfectly - we have used this since the start of GitHub platform and it is essential part of working together on GitHub. There are good tutorials on the web for coming familar with GitHub and git.

I never have had used graphical applications to manage git. As you wrote something about "gitkraken" I gave it a try. But - as I expected - working with it is destructive. I click gitkraken to start - then I change to my command line and do much stuff on command line before the splash screen of gitkraken disappears and normal window comes up :) It is very, very slow, because of it does much stuff until you enter one single command. It uses much ressources on PC and it is not reliable as it should be. Working on console is the best solution - I stay on this opinion and gitkraken has not changed anything. It left my HDD after one try of using it...

I think git is like coding. There are GUIs which creates code from dragging boxes to a "programming area window". The result is that "hello world" measures 3MB instead of 1KB and you need 5 times longer to create "hello world" using a gui than programming it using a simple text editor.