df8oe/UHSDR #1129: Ensure STM32F watchdog is correctly configured

db4ple, 2017-10-14:
I have been fight quite a bit with my mcHF board due to HW watchdog being enabled. This killed openocd uploads (watchdog did reset during the job). So it is not just disabling the watchdog in the bootloader but correctly configuring it in the option bytes (WDG_SW == on). We should check that. I think by default the watchdog is configured like this but if not, you are in trouble since you will not be able to boot before the watchdog bites back.

This issue is a reminder to find a proper solution. It is not immediate, since most will never encounter this problem due as it requires changes to the STM32Fxx option bytes.

sp9bsl, 2017-10-14:
By default STM32F4xx has this functionality disabled in option bytes. I mounted them a lot and never found it enabled by default. It is more safe to enable it by firmware at the beginning than use option bytes i think.

db4ple, 2017-10-14:
I know, it is by default off. But if one, like did today, manages to turn this "on" without noticing, it is a major pain to find out. I believe this cannot happen in normal operation but you never know, what breaks for instance if the firmware flash is interrupted or you flash over your option bytes with DfuSeDemo...

As I said, I believe this has low urgency since it is probably a very rare event.

closed 2018-06-09 by df8oe