df8oe/UHSDR #1131: Keyer memories improvements

ve3ids, 2017-10-15:
The 3 memories that are implemented are working well. Storing them to the EPROM to save during power down would be great. Also, having a repeat function for the memories with a programmable pause included would be a good benefit as well for contest operations.  (Beacon)
Thanks for the hard work. 
73 Don ve3ids

9a6cw, 2017-10-17:
finally excellent feature !! thank you very much ! It works ! two memories for different cq are enough..
73 Ivica / 9a6cw

S52CQ, 2017-10-21:
Hi, I am testig CW memory keyer. It does work great, 3 memories are more than enough. Is that a development feature that memories are not stored in the EEPROM?
73 de Jure, S52CQ

db4ple, 2017-10-21:
Yes, for now we do not implement the EEPROM but we will.

s53dz, 2017-10-22:
An excellent job done with CW memories implementation and straight forward procedures to use them.

TNX 73 Bojan

S52CQ, 2017-11-29:
During WW contest last weekend I was using mcHF CW keyer memories with great success however I have found a bug. 
When in CW mode all works well, although it's rather annoying that after keying is over TX still "hangs" for 100 or 200 mseconds, while in a contest one would require immediate switch to RX. This is not the bug - that is an observation.

Real bug is this: set mode to LSB, press play of any of memory buttons. LCD would display > sign, but nothing happens (as supposed). You can do that few times and few > signs will show. But if you then switch to CW and press that memory button all previously pressed memories are transmitted - without possibility to brake it.

This has also shown that it would be great to have one button to immediately stop transmitting memory. That would be handy in case that one presses two times same button unintentionally. Trust me - unintentional press does happen especially under 48 hours contest...

As a matter of fact there is one button still free isn't it?

73 de Jure

phaethon, 2018-02-09:
@s52cq sending PR in a moment fixing the issue with mode change and sending previously sent memories.

phaethon, 2018-02-09:
@S52CQ button is reserved for potential use. Removing those 100-200 msec after transmitting might not be trivial change in a code. I would like to postpone for later.
I suggest renaming the issue into "Saving keyer memories into EEPROM" to make it more specific. The rest has been addressed in one way or other.

phaethon, 2018-03-11:
Since today, memories are saved to EEPROM.

S52CQ, 2018-03-12:
Great!! I have tested it and it works - thanks for that! There is however one thing to check: When I press REC - and do no keying (I was presuming that this will delete the content) and then press REC again (try to get out of REC mode), I cannot exit it. 
Can you please check this? My settings:
HARDWARE	
UI-board	RF-board: mcHF v0.6	mcHF 0.6	
Display HY28A Parallel
Touchscreen: YES

DEMODULATION MODE: CW	
Filter-BW	1.8 LP 
Noise reduction OFF	
Autonotch OFF
DISPLAY	Magnify size 1x	 waterfall 
EEPROM: 24xx1026/CM01/128k
Flash size: 1024
RAM size: 192
Firmware: D2.9.14
Build: Mar. 11. 2018
Bootloader: Monka

73 de Jure, S52CQ