df8oe/UHSDR #1133: Builds just a short .bin

s53dz, 2017-10-17:
FW 2.5.122, from 17.10.2017
F4
CLI
make (all, bootloader)

Make returns just a 144 bytes long binary. Same for bootloader and firmware builds.

73 Bojan

db4ple, 2017-10-17:
Thanks! So we shrunk the code a little more than expected.  I did not check the size after my make changes since build process looked good, with all files being compiled.

s53dz, 2017-10-17:
Danilo,

I am good with 144 bytes, as long everything works, hi.

73 Bojan

df8oe, 2017-10-17:
All object files are built - but not linked together...

db4ple, 2017-10-17:
Yes, I already found my mistake. Will correct soon.

s53dz, 2017-10-17:
FW now builds ok, but not so the BL.

db4ple, 2017-10-17:
You are right. But luckily no longer. If Andreas would have been right, the fix would have worked, but for the bootloader we did not build all the required files...

closed 2017-10-17 by s53dz