df8oe/UHSDR #1137: Could you support firmware updates using the LVFS?

hughsie, 2017-10-19:
Hi, and first sorry for the issue, I wasn't sure which one of you to email, I hope this is okay to open. We have developed a tool for updating device firmware when using Linux -- we've had a large number of hardware vendors switch to this service in the last 18 months. See https://fwupd.org/ for more details.

fwupd allows vendors to either use the existing upgrade mechanisms (e.g. DFU), or provide a custom plugin that can be used to install the update. The plugin can also be used as a way to query the current firmware version, and the LVFS site allows vendors to easily submit firmware files for mirroring and signing.

Using LVFS might mean you're able to offer firmware updates onto lots of devices without having to explain how to update on Linux. It's a completely free service, and I'm happy to help you as much as you need.

There's a draft technical whitepaper here if you'd like more details: https://people.freedesktop.org/~hughsient/temp/LVFS_Whitepaper.pdf

If this isn't something you're interested in, please just close this issue. Thanks!

db4ple, 2018-06-01:
@df8oe : Please close. Don't think this is relevant for us.

hughsie, 2018-06-01:
Ohh, sorry. I thought you provided firmware updates to end users running Linux?

db4ple, 2018-06-01:
The device in question is running standalone ham radio transmitter and receiver and is only optionally connected to a PC, most of them Windows as far as we can see. Firmware update is already easy to do, and I don't think we want automated updates. So it would a niche solution... And it seems, none of the developers or users is interested in that topic. We'll reopen this, we someone wants this.

hughsie, 2018-06-01:
Okay, understood, thanks for the explanation.

closed 2018-06-01 by hughsie