df8oe/UHSDR #1138: Firmware question

WD8BXS, 2017-10-20:
Is there a place that will tell us what is changed in each new firmware update??

If there is , I do not find it.
I do know of the auto generated menu document, but that doesn't really do what I am asking.

Something like: this version adds etc...

Thanks, Chuck WD8BXS

df8oe, 2017-10-20:
Hi Chuck,

there is a short comment at each commit. You can find them here:
https://github.com/df8oe/UHSDR/commits/active-devel
Just click on a commit and you can see the comment which usually tells you what was changed by this commit.

73
Andreas

WD8BXS, 2017-10-20:
Chuck

db4ple, 2017-10-20:
Hi, I took the liberty to put Q and A into the FAQ section of the Wiki: https://github.com/df8oe/UHSDR/wiki/Firmware-FAQ

closed 2017-10-22 by WD8BXS