df8oe/UHSDR #1141: Bug of Receive audio mute

gmenga1, 2017-10-21:
starting from September 11 2017 several users of mcHF have noted that sometime the audio becomes mute without apparent reasons.
The situation is: SSB mode, spectrum still present, usually after a cycling from transmit to receive, but not always. The only cure is restarting the TRX, however I noted a brief noise in the up to then mute speaker just before the power went off, as if the audio communication was at that time again re-established.  
The last time it happened to me was with board V. 0.6  bootloader 3.4.1, firmware 2.5.122.  
It was hypothesized that the AGC system was responsible for that, especially the option  Standard AGC, but after changing the configuration  to WDSP AGC, and checking with others that the parameters in that configuration were well proved the problem resurfaced again.
A tread on this subject is active on the M0NKA  mcHF forum.
Giuseppe IZ1GCL

s53dz, 2017-10-21:
I remember some time ago having a similar issue when by testing mcHF I forgot to open the PS current-limit and as a result at transmit and at a transient of TX to RX the supply voltage temporally lowered.

But I can't tell if this is the same issue.

73 Bojan

df8oe, 2017-10-21:
I never have had this so let us try to appoach.

If the issue occurs:
Can you always listen to USB audio or is it muted, too?

73
Andreas

df8oe, 2017-10-21:
Additional: it may depend on a personal usage profile which differs from most usage profiles. So pse tell:
1) what filter do you use? Does it occur on different filters?
2)what magnify mode do you use?
3) are you running in spectrum (light/normal) or waterfall?
4)do you use compressor?
5) do you have activated any DSP function (NR, NB, peak, notch manual/automatic)?

73
Andreas

WM0I, 2017-10-28:
I also have this problem but with more information to help with this. Here are things I have seen.

1. Sometimes when powering up the audio is muted.
2. The audio will become muted when I come out of tuning up.
3. Audio gets muted sometimes when changing band, also other times the audio will be so loud for about a minute and then go back down to normal.
4. Then there is the muting for no reason when the SDR is just sitting there not being touched.
5. coming out of PTT audio will be muted sometimes.

Spectrum is in light mode.
AGC is in WDSP.
Power is at 5 watts.
Can be in NEW NOISE BLANKER or not.

I hope this helps.

WM0I

gmenga1, 2017-10-28:
after opening the issue, one week ago, and close monitoring the situation to answer to the requests of Andreas, the problem didn't occurred anymore.
I only have a suspect: a sudden reduction of voltage from the power supply.
Before the remark of Bojan, that a limitation of current in the power supply caused him a similar problem, my experiments were done with an ammeter in series with the power supply, without paying much attention to the connections. That could have been the cause, After, I disconnected the ammeter and gave to the mcHF always a solid power.
I have also to mention that in some experiments I tried to re-create artificially the problem, limiting the current from the supply, obviously without reaching the point of spontaneous switch off of the TRX. I was able to froze entirely the device (LCD frozen no answer to any command), but never to obtain to mute the only audio.
Giuseppe
IZ1GCL

WM0I, 2017-10-31:
Also if the PA is set to full power the muting occurs. If I set the PA to 5 watts the muting seems to stop so far.

Again I hope this helps

df8oe, 2017-10-31:
That can be a hint for a voltage supply problem, too. If you are running at high power and the inner resistant of your power supply is too high the voltage will drop in spikes (corresponding to drawn current). This effect is bigger at high currents than at low ones...

73
Andreas

WD8BXS, 2017-11-06:
I have one here doing the audio mute after keying a few times.. It is not the supply for sure. I will try touchin up solder around codec.

WD8BXS, 2017-11-08:
Just confirmed that when the audio goes away the USB audio goes away as well.
Also you hear a sharp increase in audio just before it goes away, Like the agc is off. 
I am using latest build.

Have re-soldered all over the codec section but to no avail!

Powered from Astron 35 amp supply.

Chuck WD8BXS

df8oe, 2017-11-08:
Hi Chuck,

very interesting. It sounds that it is a complex relationship between received signal and AGC. If waterfall / scope is already present audio is digitized and in STM present. If USB audio is lost, too it must occur inside STM - not in codec.

Please try next time to play around with all AGC settings if you can bring audio back. I never have had this combination here and actually I do not have any idea what it can be beside AGC.

73
Andreas

WM0I, 2017-12-04:
I found if you set TX Initial Muting Time in the configuration menu to about 150ms the muting and full audio blasting stops.

WM0I

df8oe, 2018-01-22:
Rich has investigated that reducing I2C bus speeds to very, very low 60KHz solves the issue at chinese mcHF clones. Maybe a hint for all which do have muting problems...

closed 2018-01-30 by df8oe