df8oe/UHSDR #1146: Frequency tuning indicator question

WD8BXS, 2017-10-22:
I under stand about the -12 khz translation, but I have seen some videos showing the line in the center of the screen.

Did I miss something??
Is there a way to get the line in the center of the screen with the translation on??

If not , why not??
Thanks,
Chuck WD8BXS

db4ple, 2017-10-22:
Magnify ! Or Zoom (same thing)

WD8BXS, 2017-10-22:
Magnify, or Zoom has nothing to do with my question!
I am talking about the tuning line at which the received signal is tuner, it is to the right side of the screen when –12khz translation is turned on. 
It would be nice to have it in the center of the screen as with all other radios.
Chuck

V31KC, 2017-10-22:
I think what he means is set your Display Magnify to 2X. that puts the red line in the center for me.

df8oe, 2017-10-22:
It is physically impossible to put tuning indicator in the middle when you look at complete received bandwidth **because of you are not receiving in the middle**. "Middle" is NON TRANSLATED - you are working in translated mode so you are "off" middle.

If you want to put it again in the middle you must fetch a saw and cut of "the longer leg". This can easier be done by magnifying. You lost some view but your marker is in the middle.

73
Andreas

WD8BXS, 2017-10-22:
Thank you for being so candid.
73 my friend,
Chuck

closed 2017-10-24 by WD8BXS