df8oe/UHSDR #1147: Very high receiver noise even when no antenna is connected.

WM0I, 2017-10-23:
The noise coming from all bands is high so much so as I am able to hear most signals that I hear from my Icom IC-718 HF transceiver. Also there is a problem with a high burst of noise when I let up the PTT switch. This burst will slowly go back down to the level before I pressed the PTT switch. I installed all the firmware so far and is up to date as of 10-23-2017. 

WM0I

DG9BFC, 2017-10-23:
without antenna the agc cranks up the gain to max ... so you hear more noise as you would with "normal working conditions" (means with an antenna you will not notice that much noise) ...
what mode did you test it with (ssb??? am??) ... what filter did you set?? what were your gain settings in menue (max rx gain etc) ... all that influences the "noise" that you hear ... 
but when you connect an antenna the agc is turned down ... and the "internal noise from the mchf itself" ... is masked from the antenna signals
dg9bfc sigi

s53dz, 2017-10-24:
Hi,

Yes, as Sigi said, it is AGC in action.

To be sure that there is no HW fault I suggest you to do:
1. Turn AGC off temporarily and try changing RF gain manualy,
2. Turn S-meter display to show dBm and check the measured RF input level. With an input closed with 50 Ohm, this should be of a magnitude from -125 to -130 dBm for CW BW.

73 Bojan

WM0I, 2017-10-24:
Sorry everyone, It has been about 10 years ago since I had a HF rig and I forgot about the AGC. I turned it from automatic to manual and set it to 25. That cleared up what I thought was a problem.  Now I am looking for the PULL-REQUEST.

closed 2017-10-24 by WM0I