df8oe/UHSDR #1149: Please add it to the new software.

sp9kn, 2017-10-24:
Please add:
-  reversing the direction of the encoder
- reversing Waterfal
- decode Cw / PSK / RTTY in Waterfal window?

73 de SP9KN

yo2ldk, 2017-10-24:
subscribe for reverse of waterfall !
  
   73 de yo2ldk!

sp9bsl, 2017-10-24:
i will add waterfal reverse, it is a must in new lcd firmware (spectrum and waterfall simultaneously)

DG9BFC, 2017-10-24:
we already have cw and rtty (one textline only but it works!!)... and i am guessing that bpsk decoding will follow soon .... (RIGHT?!?)
so only thing is reverse of waterfall and encoder
dg9bfc sigi

sp9kn, 2017-10-24:
Yes, but I'm talking about DECODING in this big window.

df8oe, 2017-10-24:
And where leave important waterfall/scope??

sp9kn, 2017-10-24:
The menu can be switched on and off for digital modes.

 

Pzremek

df8oe, 2017-10-24:
Please DO NOT QUOTE. Everyone can read the text a few lines above and quoting makes postings much mor complicated to read.

I by myself will never switch off waterfall or scope so I will not implement this. But we are already thinking about different LCD layout with more space for digital modes AND WF/scope in parallel.

DG9BFC, 2017-10-24:
when you do not monitor a band but doing a qso in a digital mode there is no need for seeing spectrum or waterfall of the whole sdampling bandwidth ... you could "lock" vfo to your qso partner, switch of the spectrum/waterfall and have LOTS of space for text (or a picture in sstv?!?) ...
i would like it to have the room available (switchable ... maybe only possible in combination with a locked vfo?? ...)
no digimode does decode while turning the dial ... so when you have found a qso partner (or are on a free spot to call cq in a digi mode) ... lock the vfo and then you can switch off the screen for your text (or the received text from your qso partner)
in my view WHILE doing a qso the waterfall or spectrum is not needed ... or just the zoomed spectrum of your filterpassband (same that you would see in a digital modes software ... 0-3kc)
...
so .. .there is a way to free up the screen for text (not an easy task cause we then have to manage two screen layouts ... maybe more depending of the more modes that could follow)
we just have to discuss HOW to do it ... with what button combination ?!?
you can already switch off the lower screen (i sometimes use it for lowering the noise a bit if i have some helicopter noise) ... why not use that area for textlines?!?
we now have only one row for received text (be it cw or rtty) ...
if we could use that lower screen area for more text lines .... that would be nice (personal view)
(what do others think?!?)
greetz
sigi
ps ... with a bigger screen and higher resolution all those changes would be easier i guess so sooner or later we have all to switch over to a bigger screen and the 429 cpu

g7jiq, 2017-10-28:
I think making the waterfall switchable on/off would be a good idea, as I don't use it myself, but some do.
Switching it off would lower background helicopter noise and also smooth out odd delays and stutters as a bit more processor power could be handed back to the main running.

sp9kn, 2017-10-28:
I think so too.

Przemek SP9KN

db4ple, 2017-10-28:
No need for development: You can switch off the waterfall or scope by setting the respective speed to zero (0).

g7jiq, 2017-10-28:
Didn't know that, thanks for the info

db4ple, 2017-10-29:
Please close the issue now, thank you!

DG9BFC, 2017-10-30:
ok then let me do a quetion:
Can you add that when user switches lower screen off that then the free area is used for received text ?? Instead the one line we would then have sevral lines for text
that would be something new in screen layout (more textlines)  without needing too much cpu load and without complete rearrange of the screen
setting waterfall speed to zero is not an option for me cause sometime i use spectrum and sometimes waterfall ... but when i am monitoring a cw or rtty station ... then i do not need spectrum or waterfall anymore but i would like to have more text lines (what do ot

sp9kn, 2017-10-31:
When the firmware will support the direction of the encoders.

df8oe, 2017-10-31:
I will not do that. II do not see an advantage for the mass of the users. If you take encoders that have other pinout as the recommended ones you only have to swap two wires. I see it as an disadvantage for most users to have one configuration point additionally.

Regarding waterfall direction:
It is functional 100% working - just "different from normal". I like things which are different from normal ;) So I will not change that, too. It is "not done in 5 minutes" because you have to swap frequency ruler upwards (otherwise you do not see newest frequency as exact as possible). You have to swap bar where you can elect magnify and wf/scope, you have to modify touchscreen operations. Wasting of time in my opinion.

But maybe there is someone who fetches these two points.

If not, issue can be closed.

73
Andreas

sp9kn, 2017-10-31:
Sorry Andreas do not waste time just do not want you. You only do what you think is right. The menu has so many unnecessary options that one more can be. Since we users help you to improve soft then can do it for us?

Wysłane z iPhone'a

> Wiadomość napisana przez DF8OE <notifications@github.com> w dniu 31.10.2017, o godz. 15:58:
> 
> Regarding waterfall direction:
> It is functional 100% working - just "different from normal". I like things which are different from normal ;) So I will not change that, too. It is "not done in 5 minutes" because you have to swap frequency ruler upwards (otherwise you do not see newest frequency as exact as possible). You have to swap bar where you can elect magnify and wf/scope, you have to modify touchscreen operations. Wasting of time in my opinion.
> 
> But maybe there is someone who fetches these two points.
> 
> If not, issue can be closed.
>

df8oe, 2017-11-27:
yes - this is a new enhancement wish...

closed 2017-11-27 by df8oe