df8oe/UHSDR #1151: Pulseaudio - Question

s53dz, 2017-10-25:
Hi,

I am using **mcHF** with **WSJT-X** over USB cable. It works just great. Tnx for implementing the CAT and audio over USB. I had to do the configuration as it is pointed out in Wiki:
```
load-module module-alsa-source device=plughw:1,1
load-module module-alsa-sink device=plughw:1,0
```
to work.

Now, if I want to use an RTL SDR stick `Realtek RTL2838` on the same machine, I need to remove those two lines to be able to use Pulseaudio for the RTL.
Is there any other solution to use Pulseaudio for both?

TNX 73 Bojan

df8oe, 2017-10-26:
Why do you need to remove the two lines? Software that works with RTL-Stick usually gives its own namings for virtual audio device.

BTW I do not need to add anything manually to modules section. But I do not use pulse audio at all for HAM usage - all programs I am using (fldigi, WSJT-X) do speak "directly" to the mcHF. I only must choose the correct device in settings and all works flawlessly. I have already tested plugging in two mcHFs and I can use one for WSJT-X and the other for BPSK / fldigi... I am nearly sure you can define a loop device where you connect mcHF audio to pulse audio - manually or via JACK. But I like it KISS so I will not try to use pulse audio...

73
Andreas

s53dz, 2017-10-26:
I am using a gqrx from OZ9AEC with RTL SDR stick.

73 Bojan

df8oe, 2017-10-26:
I am using GQRX, too. I do not need to add or delete anything. I can run GQRX and WSjTX on mcHF simultaneously - but I am not using pulse audio for that.

73
Andreas

s53dz, 2017-10-26:
Andreas,

hm, ... please, tell me about it. I suppose it is not only a version of linux?
With Realtek USB I have no troubles to set it in GQRX, but for WSJT-X to recognize mcHF audio/USB input/output I need to add those two lines in pulse/default.pa. And GQRX uses pulseaudio:
http://oz9aec.net/radios/gnu-radio/gqrx-and-pulseaudio
An excerpt:
"... since pulseaudio is the way I want to go with gqrx."

73 Bojan

df8oe, 2017-10-26:
Hi Bojan

I do not need anything for working mcHF with WSJTx and/or fldigi. I just had to activate "plughw:<recent device>" in the settings and both work without the need of any additional modules, pulseaudio etc.

I am using Linux Mint Debian Edition or Siduction (both 64bit versions). I left Ubuntu ~10 years ago because of they are doing things which are more and more incompatible with the rest of Linux world, integrate unfree and pre-alpha-staged drivers and so on.

73
Andreas

db4ple, 2017-10-26:
The Wiki says Ubuntu 16.04 for the rules and I can confirm it was necessary for 16.04 but is not required for 16.10 and 17.04 (at least on my Ubuntu machine). Neither was it required before 16.04

s53dz, 2017-10-27:
> ... please, tell me about it. I suppose it is not only a version of linux?

TNX Andreas, TNX Danilo,
Yes, **it is** only about linux version!

73 Bojan

closed 2017-10-27 by s53dz