df8oe/UHSDR #1155: Need help in doing the recommended Modifications.

WM0I, 2017-10-28:
I am 72 years old and need for someone to do the recommended  Modifications. My eye sight is in a bite of decline. Hands are not as steady as they used to be either. I am hoping there is a service out there somewhere that will do these Modifications.

WM0I

df8oe, 2017-10-28:
Chuck WD8BXS is doing a great job in building anfd servicing mcHF for OMs in USA. I think he can help you.

73
Andreas

WM0I, 2017-10-29:
Chuck is no longer working with the McHF group because of the McHF clones. Hope there is someone else.

df8oe, 2017-10-29:
If you do not own a clone there is no problem. Chuck offers help.

73
Andreas

WD8BXS, 2017-10-29:
WM0I,

Richard,
 I do NOT know where you are getting your information!!!
I certainly am working with the mcHF radio,
 I just finished building my 15th radio, and I plan to build more when they become available.

I am here to help anyone with their building of the McHF from Chris.

BUT,  I will NOT work on a clone,

De WD8BXS, Chuck

df8oe, 2017-11-01:
Please close issue if answer is satisfying.

closed 2017-11-02 by WM0I