df8oe/UHSDR #1159: Frequency display flickering red on 3.5mhz when tuned

g7jiq, 2017-10-29:
I have just updated the software to 130 and the frequency display flashes red when tuned.
other bands seem ok.
I have reverted back to 129 and all is well again.
Is there something in my menu setting set incorrect.

db4ple, 2017-10-30:
Hi,
which is the other color? I guess red: this indicates a communication problem. See https://github.com/df8oe/UHSDR/wiki/Operating-Manual#frequency-display-explained

Interesting thing is that nothing related to the Si570 or I2C was changed between 2.5.129 and 2.5.130 (we renamed the files but this does not change the functions inside). 

Please check your I2C bus speed for bus 1 (configuration menu) if it differs from the default 100 khz. 

In any case you should give the 2.5.130 another go and see if the problem persists. It is most likely related to an hardware issue (without completely ruling out a software problem).

73
Danilo

g7jiq, 2017-10-30:
A very strange fault, have tried 2.5.130 again, then back to 2.5.129 which solves the problem.

Having had a closer look  it seems all the ham bands flicker and the tuning is slightly delayed, the frequency display blinks from white to red when tuning is turned and back to white when stopped,
Any other band 6, 11,15mhz shortwave work fine,
Both i2c1 and i2c2 bus are set to 100khz

df8oe, 2017-11-01:
Do you have fixed it? It must be a hardware problem because all of my mcHFs do not have this behaviour... If answer is satisfying please close issue

closed 2017-11-01 by g7jiq