df8oe/UHSDR #1160: Information and details of firmware builds. just a question.

WM0I, 2017-10-31:
I update the firm ware every time a new firmware version comes out. After I download it and install it I go looking for information of what is in the update. I keep looking but I am not finding much. Where do I find the details of the update?

df8oe, 2017-11-01:
You can find informations about every commit here:

https://github.com/df8oe/UHSDR/commits/active-devel

73
Andreas

DD4WH, 2017-11-01:
Please also note that the newest ("daily build"/"bleeding edge") firmware versions are of an experimental nature and can have very reduced or no documentation of new features. They are also likely to have bugs and/or new (often undocumented) menu entries.
If you like to have extended documentation on implemented features, please stick to the official releases of the firmware: https://github.com/df8oe/UHSDR/releases

73 Frank

df8oe, 2017-11-03:
Sadly because of gigantic development speed the "official releases" are mostly outdated and missing many new features. So I recommend to use the "daily builds/bleeding edges".

It is like using bleeding edge Linux. I am working with Debian unstable (!!) (which mostls is more stable than "Windows stable") and often there is no documentation - you have to do try and experiments. But that is amateur radio: EXPERIMENTAL :)

df8oe, 2017-11-04:
If your question is answered satisfying please close issue.

closed 2017-11-04 by WM0I