df8oe/UHSDR #1165: Rf Gain in System info showing error!

Derek1965, 2017-11-04:
Rf Gain in System info showing error!

Derek1965, 2017-11-04:
Rf Gain in Configure menu showing error!. Regards Derek.

DD4WH, 2017-11-04:
Hi Derek,
1. try the newest version, you seem to have a version which existed only 10 min and contained a small bug with Rf Gain
2. Where can you see Rf Gain in System Info??

df8oe, 2017-11-04:
Thanks for reporting. We have noticed that already and fixed it. "Max RF Gain" was an obsolete entry of old AGC which is gone now.

If it is ok for you now pse close issue.

73
Andreas

DD4WH, 2017-11-04:
Hi Derek, hi Andreas,
our comments crossed ;-)
Please Derek, could you try the very newest version and report whether the issue has disappeared or not?
Thanks in advance!
73 Frank

Derek1965, 2017-11-04:
Just loaded 2.5.133 issue has now gone, Thank you everybody for your hard work. Derek

closed 2017-11-04 by Derek1965