df8oe/UHSDR #1169: Error Code 4

Tibet4690, 2017-11-06:
I am getting Error Code 4 when trying to upgrade to the latest firmware [2.5.134]. I am currently running 2.5.130.

I am attempting to use a Kingston Data Traveler which is USB 3.0 with a capacity of 32GB.

Any suggestions?

df8oe, 2017-11-06:
Stick damaged, USB hardware error (bad solderings), 5V on USB plug instable or too low. And possibly try another stick. We do not have had one single stick which is "incompatible" - but you never know...

73
Andreas

Tibet4690, 2017-11-06:
Thanks for the info, I must have used a 2.0 stick without realizing it previously!

df8oe, 2017-11-06:
Please close issue if this solves your problem.

73
Andreas

Tibet4690, 2017-11-06:
Ended up using the DFU method as I did not have a USB 2.0 stick available. Worked as expected.

Thanks again for the help and will be sure to mark closed.

closed 2017-11-06 by Tibet4690

db4ple, 2017-11-06:
Hi, you don't need a stick with UBS 2.0, but it has to be formatted using FAT (not exFAT, not NTFS). We will update the bootloader documentation accordingly.