df8oe/UHSDR #1173: USB / Wireless / interface?

WM0I, 2017-11-08:
Is anyone working on doing firmware to get a keyboard interface and the software to use with it?

df8oe, 2017-11-08:
Keyboard support via USB is already implemented. Actually it is used for RTTY and BPSK transmitting. Wireless keyboards with USB dongle are supported, too. BUT: only devices with ONE configuration can be used.

73
Andreas

WD8BXS, 2017-11-09:
I got one of the small wireless keyboards that were recommended, it works, but my simple small usb keyboard that I use on my icom  does not.

Chuck WD8BXS

WM0I, 2017-11-09:
Hi Chuck, which keyboard is that?

Richard WM0I

WD8BXS, 2017-11-09:
Hi Richard,

I got this one:  https://www.ebay.com/itm/Rii-Mini-X1-2-4G-Wireless-Air-Keyboard-with-Mouse-Touchpad-NEW/253208748356?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Chuck

From: Richard McKinney 
Sent: Wednesday, November 08, 2017 9:28 PM
To: df8oe/UHSDR 
Cc: WD8BXS ; Comment 
Subject: Re: [df8oe/UHSDR] USB / Wireless / interface for a keyboard? (#1173)

Hi Chuck, which keyboard is that?

Richard WM0I

closed 2017-11-10 by WM0I