df8oe/UHSDR #1174: TR4W CW CAT

S52CQ, 2017-11-08:
Hi,
I've noticed in last SW versions that you have implemented PTT via virtual RTS  so I have performed a test.

I am using TR4W loggig SW at CW mode. I have set PTT via virtual RTS=ON on mcHF.

In the TR4W programme I have set: 
Radio One=FT817
Radio One Control Port=COM3 (STM Virtual COM port)
Radio One CAT RTS=ON
Radio One CAT DTR=ON
Radio One BAUD=9600

Keyer Radio One Output Port= COM3
Radio One Keyer RTS=PTT
Radio One Keyer DTR=CW

It appears that this setting works well, PC keys mcHF in CW mode nicely however, it does not go back to RX. It hangs in TX mode (no output though).

I have to set Radio One Keyer RTS=OFF, then mcHF would go to RX, but of course then I cannot key it via PC. So the I have to set Radio One Keyer RTS=PTT and PC can key mcHF but it would not return to RX.

73s de Jure

df8oe, 2017-11-09:
PTT via RTS is already reported as "working"... I do not have any software that uses RTS or DTS - all my software uses CAT commands. So we must wait if someone can confirm that it is working (or not).

73
Andreas

S52CQ, 2017-11-14:
Hi Andreas, 

I can be your hands for testing...

I have figured it out how it works.

In the RX mode I have set PTT via virtual RTS=ON on mcHF and mcHF goes to TX mode. Then I started keying with TR4W SW, all works well, there is output present, I can hear monitor tone while keying.
But when I stop keying mcHF does not go to RX - which it would go if I manually set PTT via virtual RTS=OFF on mcHF.
After that mcHF goes to RX normally and it stays there. So the SW should monitor if keying is present and in this case turn RTS=ON

Hope that helps.

73 de Jure

db4ple, 2017-11-14:
Hi Jure,
I don't understand what you are saying: If you enable the  "Virtual RTS controls TX"  does the mcHF go immediately to TX or only when you start TXing with TR4W? 
If it goes immediately to TX, TR4W does not handle RTS the way it should handle it: It seems to have the  RTS signal always set active and in this case it is absolutely normal that the mcHF never stops sending, since this is exactly what Virtual RTS is supposed to do (RTS set to active-> switch to TX).
In the other case, we may have to look for a bug in the software.

73
Danilo

S52CQ, 2017-11-14:
Hi Danilo,

yes, you're right, when I start TR4W and set RTS =ON then mcHF goes to TX (without keying). But if I then set RTS=OFF mcHF still remains in TX. Is that what it's supposed to be?

73s de Jure

db4ple, 2017-11-14:
No, it should not continue transmitting. Independent from that, it is very likely that the RTS handling of TR4W is not okay.

Now I want to focus on the problem that the mcHF remains in TX.

Please provide me with a complete set of instructions how to repeat the problem. 
I.e. tell me which mode you are in (CW, SSB, ... ). Do I have to run the TR4W and with which settings.
Turn off the mcHF and describe the minimal set of steps in order to provoke the problem in the menu (i.e. that after switching RTS on and off, it remains in TX). Question is, do I have to transmit, or is it sufficient to go straight after boot to the menu and problem occurs.
73
Danilo

S52CQ, 2017-11-14:
Hi Danilo,

OK, let's go from the start, no TR4W started:
1. turn on mcHF in CW-U mode
2. selecting RTS=ON or OFF does not make mcHF to go to TX
3. select RTS=OFF
4. Start TR4W
5. Settings of TR4W are described at start of this issue
6. select a contest
7. while RTS=OFF no keying is possible from TR4W
8. turn RTS=ON>> mcHF goes immediately to TX, but no keying
9. start keying with TR4W - there is CW RF signal and one can hear side tone; when keying is stopped mcHF remains in TX
10. set RTS=OFF>> mcHF goes to RX
11. now repeat point 8>> mcHF goes to TX
12. set RTS=OFF>> mcHF remains in TX until TR4W makes some keying, then repeat point 10 and mcHF goes to RX

Hope this helps.

73s de Jure

S52CQ, 2017-11-27:
Hi Danilo,

was my description enough detailed? Do you need more info on that to replicate a problem?

73s de Jure

df8oe, 2017-11-27:
Hi Bojan,

this is an issue in TR4W.

Please test as following:
- use fldigi
- in "config/radio settings/hamlib" unmark "PTT via hamlib command"
- in "hardware PTT" mark "seperate serial port for PTT", use RTS", and do NOT mark "RTS=+V"
- in mcHF set ""PTT via virtual RTS" to ON

run this config. You can see that rig transmits if you push PTT on fldigi and immediately goes to RX if you release PTT.

Maybe the names in English are different - I am running a German localized version of fldigi...

vy 73
Andreas

df8oe, 2017-11-27:
Additional investigations:

"PTT via virtual RTS" was implemented because UCXlog wants two different signals in CW: one for "preparing for TX" and one for "keying" (what a sick idea...) That was implemented in the way "preparing to TX" and "keying" are done via CAT command and virtual RTS. TR4W also wants two different signals - but sadly both on RS232 lines (and different ones). To work with that program there must be added a new feature that can enable "keying via signal 1" and "PTT via signal 2" where the two signals must each seperately selected from "DTR/RTS/CAT" AND both are forbidden identical. This is the first time in my life I have seen such an implementation...

73
Andreas

S52CQ, 2017-11-27:
Hi Andreas,

maybe it's an issue with TR4W, but N1MM SW that's being widely used among HAM contesters has same: keying via DTR and PTT via RTS (or keying via RTS and PTT via DTR). 

I was thinking to do it like this: when I start/stop keying the SW would make a toggle of RTS - like I do it in the menu by hands in points 8-10. 

I will try also UCXlog to see how it works there.

Best 73s de Jure

db4ple, 2017-11-27:
Hi,
it is implemented in the UHSDR firmware like this (this is how UCXLog works and which considering the FT817 hardware makes "sense"):

Virtual DTR is used for keying, Virtual RTS can be used for PTT. Support for virtual DTR is always active, but works only for CW of course AND if a computer activated PPT is used (either using the FT817 CAT command or using Virtual RTS). Virtual RTS needs to be enabled and then works similar to CAT PTT, i.e. the PTT is active until the virtual RTS lines is released.

UCXLog uses CAT for frequency control and PTT. The problem is that in the  FT817 CAT protocol there is no such thing as CW keying. The FT817 expects CW keying from the CW input port. That is Virtual DTR.
This works in our firmware.

N1MM+ has an issue with that, since it will not use the CAT interface for PTT control, only frequency control. Because of that I implemented Virtual RTS. In N1MM you control the UHSDR with just Virtual DTR and Virtual RTS. And it works, at least at the time I tested it. You can try for yourself.  

It does not work in TR4W because this program somehow never releases the Virtual RTS (aka PTT) after first transmission, at least that was my finding for now. There is not much we can or will do about it. It seems to me like a programming issue on behalf of TR4W. 

If however, N1MM+ does not work, then we have regression somewhere. See #966 for background.

n4af, 2017-11-27:
TR4W does not support SDR.

You might want to experiment with the CTRL-J settings for PTT.
Much of our radio interface can  be found at
https://github.com/n4af/TR4W/blob/master/tr4w/src/trdos/LOGRADIO.PAS

73, Howie

db4ple, 2017-11-27:
Hi Howie,

TR4W does know, we are an SDR. It thinks we are a FT817-TRX with an external DTR/RTS keyer. But you are right, someone should look into the setting possibilities of the TR4W and see if a working configuration can be found.

73
Danilo

n4af, 2017-11-27:
Well, from my standpoint the software must correctly emulate the FT817 so
that TR4W does not know the diff.

73, Howie

db4ple, 2017-11-27:
Hi Howie, that is right but not the point here. DTR and RTS are not FT817 specific but emulate a keyer, which in turn in theory would be connected to a TRX, in our case to the virtual "keyer" inside the UHSDR.
So FT817 compatibility is required for the CAT part but not the keying part in TR4W.

S52CQ, 2017-11-29:
I can confirm that mcHF works well with N1MM software in WinXP! 

It does look like the TR4W issue to me.

73s de Jure

db4ple, 2017-11-29:
So, unfortunately, this is a "Can't fix issue here". Problem is on TR4W side or in the specific interaction between TR4W and our implementation, which, however, as far as I can see, does the right thing. 

Anyone willing to debug this, raise there hands now, please. Otherwise, expect no progress. Sorry. I cannot debug TR4W, as I and probably most other UHSDR developers lack the skills and tools to compile and debug the TR4W side (it is Delphi or Pascal or both). 

73
Danilo

S52CQ, 2017-11-29:
Yes Danilo, you're right. As much as I like TR4W SW I cannot use it with mcHF. Same goes also for many mcHF users around the globe...

So Howie, if you're reading this let me hear your comment before I close this issue.

73s de Jure

n4af, 2017-11-29:
TR4W does not support software emulation of hardware, but will support/compile suggested changes to our open source code.

db4ple, 2017-11-29:
Yes, when I looked at the issue, I found that RTS is not released by TR4W. I verified that using an USB protocol analyzer and also checked that on the UHSDR firmware side with an real-time debugger. And 

Of course, that would work. This is how UCXLog does it (UHSDR Firmware uses the FT-817 protocol). And it would be the prefered way, since the "Virtual RTS" function needs to be explicitly enabled. Some CAT programs  play with RTS on the CAT interface when they operate the CAT and are then switching the TRX to TX without knowing. Normally not a problem since in most cases the RTS/DTR for CW is a separate COM port. For that reason, operator has to enabled the virtual RTS function. 

Fixing the RTS issue would also work, this is how the UHSDR is connected  with the N1MM/N1MM+ program (RTS/DTR).

S52CQ, 2017-12-05:
N4AF Howie [has closed](https://github.com/n4af/TR4W/issues/290#issuecomment-348814066) the issue that I've opened on the TR4W github stating that TR4W does not support that.  
That TR4W does not support that I know already but was kind a hoping that this bug will be resolved, but it seems it will not be...
If TR4W programmers won't reply to this thread I will close it with remark that TR4W is not suitable/recommended for use with mcHF/UHSDR.

db4ple, 2017-12-05:
Hi,
I was under the impression that the TR4W folks would consider looking at it. I am sorry, but I am afraid, we cannot do much here unless someone looks at it and has an idea how to deal with the issue, if at all possible, from our side. Please review my TR4W statement on the Wiki page https://github.com/df8oe/UHSDR/wiki/USB-CAT-and-AUDIO-Mode. Feel free to adjust statement.

ny4i, 2017-12-05:
It’s on the TR4W list to check out and see what needs to be fixed with RTS.

Tom NY4I
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

db4ple, 2017-12-05:
Great! We'll keep that issue open then :-)

S52CQ, 2018-01-21:
Hi Tom,

any news regarding TR4W problem with RTS?

db4ple, 2018-01-21:
Hi,
if the PTT VIA command can work with the same serial port as the DTR keying serial port, there is no reason why this should not work. We use the same approach with UCXLog and N1MM+ with success.

Danilo

ny4i, 2018-01-21:
The CAT command can go over the same serial port as the any other CAT commands you send.

Please remind me which radio you emulate and I will confirm no issues with the CAT command for PTT we use.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

db4ple, 2018-01-21:
Hi,
it is the FT-817 protocol. Don't think TR4W should have problem with this one.

Danilo

S52CQ, 2018-01-22:
Hi Tony, sure will try it and will keep you posted.

73 de Jure, S52CQ

df8oe, 2018-06-09:
Does it work now for you? If so: please close this issue.

ny4i, 2018-06-09:
I’m tracking the issue in the TR4W GitHub. It is a resource issue to get it fixed, but I hope in July along with some other pending fixes.

If you want to close it here, we will track it over there.

Tom NY4I
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

S52CQ, 2018-06-09:
OK, will close it here.

73s de Jure, S52CQ

closed 2018-06-09 by S52CQ

db4ple, 2018-11-06:
Hi Jure, Tom,
please test the newest binary. It should fix the RTS problem with TR4W and UHSDR. 
@ny4i: Mea culpa. It was our fault after all (in fact mine), everything in TR4W is working as it should. Please close the respective TR4W issue: https://github.com/n4af/TR4W/issues/290 or have it closed.

73
Danilo

S52CQ, 2018-11-06:
> close the respective TR4W issue: n4af/TR4W#290

Hi Danilo,

thanks for the info, will test binary within few days and report back.

73s de S52CQ, Jure

S52CQ, 2018-11-06:
Hi Danilo,

I have tested bin and it appears that CAT is not working here. So I wonder what are your settings of TR4W and mcHF? It must be that I have one of these settings wrongly set.

73s de Jure