df8oe/UHSDR #1175: Not working at RTC Calibration PPM

ly2bok, 2017-11-09:
Clock in SDRmcHF  working ! Thanks .
But not working at RTC Calibration PPM. No response to a plus of 450 or minus -450. Checked out.
Justas LY2BOK

s53dz, 2017-11-13:
Hi there,
I have made a quick calculation. I did not go into the code to see how it is done, I rather took these numbers as input:
RTC Q = 32768 Hz, minimal correction = 1 ppm.

Calculation for 1 ppm shows that this means a 2.592 Hz difference in 1 month. It is 0.0000791 sec in 1 month.
This is then 0.0356 sec for 450 ppm in 1 month.
This last is 0.427 sec in 1 year.

If I am right with that, you can not see the difference of +-450 ppm settings on a short hand.

73 Bojan

ly2bok, 2017-11-13:
Hi... Bojan.
OK, got it. But the micro-watch quartz has big differences and the correction according to your calculation does not produce results. For example, my inserted quartz is a day left behind for a minute. These are big differences and the correction is required for a much larger range. At least 100 ppm match for about 1 second. At least i think so I tried a few quartz, likewise far from the truth .....
Justas LY2BOK

s53dz, 2017-11-13:
Since I don't know the STM32 hw that much, is it also a STM clock XTAL in the game here?
Differences as you are stating are very large. I doubt I was just so lucky when I took **an old XTAL** from an old ARM board.

73 Bojan

ly2bok, 2017-11-13:
Perhaps you were lucky with the quartz ...
Justas LY2BOK..

db4ple, 2017-11-13:
60 Secs per Day -> 1- 86460 / 86400 = -693 * 10^-6 aka -693 ppm: You cannot correct that with +/-480ppm. 

You have probably not the right load capacitors for your quartz. I highly recommend to buy a proper quartz with known load capacitor values or even easier: buy a 32768 khz quartz with integrated capacitors. The STM32F RTC likes lower values better (single digit pF loads)

db4ple, 2017-11-13:
BTW: 24*60*60 = 86400 secs per Day.

1ppm -> 86400 / 10^6 -> 0.0864 secs per Day, i.e. 488ppm -> ~42s per Day max correction.

s53dz, 2017-11-14:
> 60 Secs per Day -> 1- 86460 / 86400 = -693 * 10^-6 aka -693 ppm:

yes   60 / (60 * 60 * 24) = 1 / (60 * 24) = 0.00069444...
I am attaching a TI document which I find interesting explaining the RTC accuracy trimming.

73 Bojan

[swca024.pdf](https://github.com/df8oe/UHSDR/files/1469861/swca024.pdf)

ly2bok, 2017-11-14:
OK. thanks.
Justas LY2BOK

closed 2017-11-14 by ly2bok

s53dz, 2017-11-17:
I have made some short-term! measurements on STM32F4 mcHF RTC. The results are in the attached table.
It seems the calibration works ok for the negative ppm values and zero but not so for the positive ones. It is showing like an integer signed/unsigned situation.

73 Bojan

[RTC-ppm-rev11-17a.pdf](https://github.com/df8oe/UHSDR/files/1482854/RTC-ppm-rev11-17a.pdf)

s53dz, 2017-11-19:
Danilo,
tnx for the fix. I am attaching my today's measurement using a +200ppm correction. It works now in the + direction as the diagram shows.

73 Bojan

[RTC-ppm-rev11-19.pdf](https://github.com/df8oe/UHSDR/files/1485386/RTC-ppm-rev11-19.pdf)