df8oe/UHSDR #1176: Strange problem

WD8BXS, 2017-11-10:
I have a MCHF here that is acting very strange. I am on 40 meters, display shows 7.192 USB, adn I am hearing stations that are on 7.168 **LSB**

Very strange, I don't know where to start with this one hihi

de Chuck WD8BXS

df8oe, 2017-11-10:
Hi Chuck...

this is not a firmware issue and you can find the issue by doing some maths :) Difference between two frequencies is 24KHz - what means that you are running in Xlate -12KNz. Your problem is loacated in section mixer / quadrature LO stage. Onde of the 90° LO signals is missing or mixer IC is dead.

73
Andreas

WD8BXS, 2017-11-10:
Great info Andreas, thanks for the quick reply!!

Chuck

From: DF8OE 
Sent: Friday, November 10, 2017 9:15 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Strange problem (#1176)

Hi Chuck...

this is not a firmware issue and you can find the issue by doing some maths :) Difference between two frequencies is 24KHz - what means that you are running in Xlate -12KNz. Your problem is loacated in section mixer / quadrature LO stage. Onde of the 90° LO signals is missing or mixer IC is dead.

73
Andreas

WD8BXS, 2017-11-10:
Good call, I found a mis- soldered resistor.  Thanks again,

Take care, and God Bless,
 Chuck WD8BXS

closed 2017-11-10 by WD8BXS

df8oe, 2017-11-10:
;)

well done, Chuck!

73
Andreas