df8oe/UHSDR #1180: Instant power on

WD8BXS, 2017-11-12:
Not sure if this is a firmware problem, but, Since new firmware 2.5.134, as soon as I plug in the power cable, the mchf turns on.

Powers down normally and stays off til powered back on, but if you unplug power and re-insert, it comes right back on.

Any ideas??

de Chuck WD8XBS

df8oe, 2017-11-13:
Hi Chuck,

this is a hardware issue.

McHF only powers up if pin2 of LM2941 gets LOW level. There are only two possibilities that this can occur:

1) via D2 (upper part)
2) via D2 (lower part)

D2 upper part is connected to GPIO of STM32. GPIO can only work if STM has its power supply - but STM does not have power supply so this line is floating and no low level can switch on 8V regulator.

D2 lower part is only connected to power button and if it is not pressed there also cannot be low level.

So LM2941 cannot power up and no 5V or 3.3V can be produced.

Question that you must answer:
where for devil is the source of low level which switches LM2941 on??

Ways to investigate is to cut tracks from the two diodes and look if issue is already present or not.

Good luck
73
Andreas

WD8BXS, 2017-11-13:
I changed over to a small 2.5 amp switching supply and it did NOT turn on when plugged in, so it must just be 
a strange issue with using the high amp supply.
Chuck

closed 2017-11-14 by df8oe

s53dz, 2017-11-17:
I would say that this can happen if there is some additional stray capacitance between the RF board U3 ON/OFF pin 2 and GND coming somewhere from the board. This would additionally pull down the pin 2 during settling the +12V input line on pin 4 of U3. Guessing.

73 Bojan