df8oe/UHSDR #1182: When keying the transmitter mcHF emits short chirp noise whith the PA at FULL power

isenburg, 2017-11-14:
When I press the PTT with the PA at full power a short chirp sound is being sent. This also happens when using digi modes via USB. If you press the PTT a few times in quick succession, the problem does not arise. It does not occur using lower power settings.

df8oe, 2017-11-14:
I am nearly sure it is a hardware issue. That can be:
- oscillating driver or PA
- influence via power supply (inner resistant of power supply and/or cables too high)

Of course there are other possibilities.

stefendy, 2017-11-24:
I also added a 100 watt homebrew PA but did not get short chirp noise or other noise in the modulation, check: wiring, PA self oscilation, LPF on PA, shielding the middle between uiboard and rfboard

df8oe, 2017-11-27:
Do you have solved your porblem? If so: please close this issue.

isenburg, 2017-11-27:
I noticed that the adjustment and configuration settings are all back to default… I will run through the setup again.

closed 2018-01-30 by df8oe