df8oe/UHSDR #1183: 40 meter buzz in transmit audio.

WM0I, 2017-11-14:
THe only band that I have this audio buzz in the transmit in is 40 meters. I have made several contacts who tell me I have RF in the mic cable. I am using a 1 to 1 balun on all my dipole antennas to keep the RF out. 40 meters is the only one affected. Power is 5 watt setting. I have set all the PA baises I think correctly.  I think what they are hearing is CPU buss noise getting into the transmit audio. Help!

s53dz, 2017-11-15:
Hi Richard,

If that is only on 40m I would guess you probably have PA self-oscillations there. I wonder how this is at only 5 W output, but anyway. Please check this Modification:
RF-04/05/06-N-009 PA Self Oscillation Prevention
on the page:
https://github.com/df8oe/UHSDR/wiki/Recommended-Modifications
and the optional R83a in PA supply.

BTW: But why would you think this is a FW issue?

73 Bojan S53DZ

df8oe, 2017-11-15:
Do you run with PIN-Diode-switching? Often this switching is the cause of distrortions and/or self oscillation, sometimes only one some bands, sometimes only on small gaps of a band... It is recommended to use a relay switching.

73
Andreas

WM0I, 2017-11-16:
I would like to replace R81 and R82 from 220 ohms to the recommended value of 100 ohms. Before I open up the case I would like to know where I am looking to find these resistors. What board and location.

Thanks in advance.

df8oe, 2017-11-16:
These resistors are on rf board (gate resistors of PA MOSFETs). But I am nearly sure your issue is the pin diode switching. Sympthoms match 100%.

s53dz, 2017-11-16:
Richard,

you can easily check if Andreas's guess also solves your trouble: 
Just remove the C78 in ANT SWITCH on RF board and see if the TX 40m signal clears up. Of course during this test there will be no receive possible.

73 Bojan

closed 2017-11-21 by WM0I