df8oe/UHSDR #1188: Bad encoder function

poporange630, 2017-11-19:
Regarding issue #734 
Symptom:
Decoders and BAND selection buttons was not triggered properly or triggered at the wrong direction.
I was about to believe that it's a decoder quality issue, until I find out that, in some cases, the BAND selection button plays up too! It's not a decoder but two simple push button!
Replication:
In some random cases, when I attempt to change band, let's say from 21mhz to 7mhz, at each press of the BAND DOWN button, the frequency change as follows: 21mhz -> 14mhz ->10mhz ->14mhz ->10mhz -> 7mhz. It looks very similar to the issue #734. Now I tend to believe this is a hidden software issue. Please let me know if I can help debugging.

df8oe, 2017-11-19:
I don't think this is a bad encoder function - I think you are confused by your own settings you have made ;)

It is possible to store any frequency on any band storage place if you enable this in configuration menu. So you can set all band storage places to 20m frequencies and if you push BAND+ or BAND- it seems that npthing changes ;)

Because of storing frequenencies outside the band storage place can cofuse that way we have implemented a second indicator which is places in white letters right hand of dBm display box. Please try again and do not look at the read band indicator (which only shows your tuned frequency on the band storage place but not the band storage place itself). I am nearly sure if you watch the white "band storage place indicator" you will see all is working find.

df8oe, 2017-11-24:
Is your problem gone? If so: please close issue.

poporange630, 2017-11-24:
Unable to replicate for now, will close the issue now and if it ever
re-appears, I'll make sure I record a video.
Thank you for your help!

closed 2017-11-26 by df8oe