df8oe/UHSDR #1192: [Question] AGC settings when operating digital mode?

Tibet4690, 2017-11-20:
This is perhaps a dumb question, but when operating digital modes what AGC setting should I be using?

I see people mentioning "slow", so does this mean setting AGC closer to a value of zero or do I need to go into my setting and change something else?

DD4WH, 2017-11-20:
Hi,
if you do not know what a slow AGC is or how the AGC in our UHSDR works, have a look at this:
https://github.com/df8oe/UHSDR/wiki/Automatic-Gain-Control-(AGC)
You will find that it can be highly rewarding to know how to adjust your AGC perfectly for your specific band condition, it can mean copy vs. no copy :-).

Concerning Digimodes and AGC, you could consult this:
https://github.com/df8oe/UHSDR/wiki/Digimodes

Specifically for RTTY & AGC you could look at this:
http://elecraft.365791.n2.nabble.com/AGC-and-RTTY-Decoding-was-K3-AGC-White-Paper-td7627781.html

73 Frank

Tibet4690, 2017-11-26:
Sorry for the delayed response, Thanksgiving can be a busy holiday! Thanks for the info!

Should I be marking an answered question as closed?

df8oe, 2017-11-27:
In principle: yes. But I do it now for you ;)

closed 2017-11-27 by df8oe