df8oe/UHSDR #1211: PTT via virtual RTS

stefendy, 2017-11-24:
When ptt via virtual rts then on external ptt pa / accesories not active, is it so or i missed a setting ?

-Bootloader 3.4.4
-Firmware 2.7.11

*Sorry for my bad english

df8oe, 2017-11-24:
Sorry - cannot understand.... You can find this menu option in configuration menu. When activated you can key mcHF via RTS line.

For keying external devices you can use the ACC connector. But optocoupler which is used in mcHF by default works very poor. Maybe you need an additional switching transistor or replace optocoupler by opto relay.

73
Andreas

stefendy, 2017-11-24:
I have activated ptt via virtual rts in configuration, and have replaced optocopuler, when ptt via virtual rts is turned on (by software) mchF has transmitting but on external ptt pin is unactivate  (not grounded) but when transmitting using ptt in mchf (uiboard J3) then on external ptt is also active (connected with ground)

df8oe, 2017-11-24:
Thank you for clarification. Now it is clear. We will take a look at it.

closed 2017-11-25 by stefendy