df8oe/UHSDR #1212: RAM problems

df8oe, 2017-11-24:
There is some RAM overlapping at 2.7.11, maybe earlier (versions after 2.7.3). I tested on OVI40: if I use CAT waterfall and scope display is corrupted. There is an issue report at amateurfunk-sulingen.de discussion board where something like this is reported for mcHF, too

df8oe, 2017-11-25:
Update: I can confirm that scope / watrfall on OVI40 no longer is corrupted at usage of CAT. Because my mcHFs all do have 256KB RAM I cannot test if there is a problem at 192KB RAM devices.... I will go back to amateurfunk-sulingen.de and discuss that with Chris, HB9BDM.

DG9BFC, 2017-11-25:
i have small ram (192k) ... sometimes spectral noise reduction works .... (and sounds fine) but most times it does not work (and produces a bad hum noise added to the signal) ... maybe also a ram problem?? (cause it does soemtimes work or not when switching on waterfall/spectrum or jumping in the menue...)

db4ple, 2017-11-25:
Hi Sigi, please verify with newest firmware or confirm problem still exists. We reorganized memory today and yesterday night. I don't have that problem on my 192k machine.

DG9BFC, 2017-11-26:
just installed 2.7.16 (newest from sunday morning) ... problem is still there ... sometime (but seldom) the noise reduction does work fine ... but most time it does not ...
can be forced to "restart audio engine" with jumping in menue (waterfall/spectrum off and back on when out of menue) ... so ... when i switch on new nr and it does not work ... pressing several times on menue and it does work ... do i have to do a full reset of the parameters?!? (did not try yet) what i did is i scrolled trough the menue points and checked them if there a value was changed ... (only new values were the ones from the new noise reduction ...)
greetz
sigi

df8oe, 2017-11-26:
My observations:

On OVI40 new NR is always working perfectly. Noise reduction is very good, distortions of audio are present, but low. NR is working under all cirsumstances (wf/scope on/off, menu opened, toggling between all combinations)

On mcHF with 256KB RAM it works most of the time perfectly. It always can be en/disabled. If I toggle between menu on/off sometimes audio gets much more distorted. It is always readable but underlayed with a crackling floor. If I en/disble menu display a few times it goes away and sometimes after toggling again it is present again. It never happens during normal operation, only when I toggle menu on/off.

I cannot test mcHF with 192KB RAM because I do not have access to one of these.

73
Andreas

DD4WH, 2017-11-26:
Hi Sigi,
try the version before the newest version (latest on Saturday), that has the spectral noise reduction include AND the lowest RAM usage. The version of today (Sunday morning) has higher RAM usage because of the added automatic notch filter (long tone elimination) ;-).
Will become better in the future, because we will save RAM when we throw out the old NR and auto-notch functions . . .
73 Frank

DG9BFC, 2017-11-26:
latest on saturday is .. aehem ... 2.7.15 ???

DG9BFC, 2017-11-26:
i have both and will try both 2.7.15 and 16

db4ple, 2017-11-26:
Hi Frank, Andreas,

RAM usage cannot have an influence in the described scenario unless there is a programming error. I rather suspect that there are out of bounds conditions in some array accesses. We need to check for that. Why doesn't make a difference on the OVI40: Not because of the more RAM but because due to different layout of the data in the RAM (on F4 we use the CCM for some data but not all) on F7 aka OVI40 we place all data in the same memory region (because there is one very big one in the F7).

So we should carefully check for arrays boundaries being obeyed in the audio code (and the display code and elsewhere of course, but audio is the most likely candidate), especially and most of all if it comes to writing into the memory.

DD4WH, 2017-11-26:
OK, thanks Danilo for your explanations! I will go and check audio_nr.c first of all, because that is the most likely candidate for out-of-bound errors at the moment . . .

db4ple, 2017-11-26:
I compiled with -O3 and at least the compiler was not able to detect any out of boundary conditions, which does not mean too much, but at least it could not find one (which it sometimes does).
I am still wondering why going in and out of the menu makes a difference. Spectrum/waterfall graphical updates are turned off (which saves some time) but apart from that not much which changes. Except of course the timing. Maybe we have a timing issue/race condition instead of a boundary overflow.

DD4WH, 2017-11-26:
A while ago I reported a very similar problem with the new noise blanker (not noise reduction) which did distort the audio on my mcHF machine, but not on others. This distortion also disappeared when I was in the menu (when waterfall/spectrum was not displayed) and reappeared when leaving the menu. The similarity between the two cases is: access to the freedv_buffer & spectrum/display not running/menu mode . . . does that help?

db4ple, 2017-11-26:
That is a hint to such a problem. It could be that it does not occur on the OVI40 due to much higher processing power available here.

DD4WH, 2017-11-26:
yes, but neither the noise blanker nor the spectral noise reduction take a lot of processing power. The requirements in that respect are very modest.

db4ple, 2017-11-26:
But that does not prevent existence of a race condition in the NR code when held up by a long running display update. It is about timing and the order of events. Not necessarily about the NR processing time alone.

df8oe, 2017-11-26:
More precise:
The event which makes the distortions appear *or* disappear is changing the state "menu on/off". A change of the state may or may not lead to a reaction regarding distortions. The direction of the changing is without influence. Reaction can be at menu on --> menu off and also menu off --> menu on. The state of distortions (yes / no) can endure more than two or three state changes of menu.

I have not investigated anything but my feeling tells me it is a race condition.

Working speed of mcHF and OVI40 is different. When I attach both to the same antenna listening to the same stations I got delays like comparing OVI40 to analog rx.

On OVI40 I never have had the distortions. They are more intensive in spectrum mode than in waterfall mode. "Intensive" menas the grade of distortion - not the frequentness.

db4ple, 2017-11-26:
You should test for buffer underruns in the audio_driver. I.e. turn on the red led if you don't get the required buffer data from the NR buffer. That kind of thing would for sure cause audible distortions.

db4ple, 2017-11-26:
That again is even more interesting. Unfortunately I have no means to reproduce this due to no RF board  connectd to my mcHF UI board.

db4ple, 2017-11-26:
@DD4WH: How have you measured the performance of your implementation (with which results)?

DD4WH, 2017-11-26:
performance of what? of the noise reduction?

db4ple, 2017-11-26:
Performance here: Timing of the actual NR function (how long does it take to process call to the actual NR function) not the functional performance (how good it does the job).

DD4WH, 2017-11-26:
@Andreas: from the software point of view, you should have exactly the same timing of the audio in the mcHF and in the OVI40, if you have the two radios running on the same station!
There is not difference in latency in the audio path. However, it could be due to different codec hardware usage, but that is beyond my knowledge . . .

DD4WH, 2017-11-26:
Timing of the noise reduction: I did not measure that.

First implementation of the NR was on my Teensy Convolution SDR: on that machine I have a very linearly programmed audio path (:-). If something takes too long, you can immediately hear it crushing in the audio. So I said: NR does not take too long, because the audio was OK. Second reasoning: the load on the Teensy increased only a few % of the total CPU load (Teensy 3.6 is a single precision FPU machine with 180MHz, so comparable to STM32F4). The same is observable on the OVI40: when noise reduction is switched on, the processor load increases by max. 1%.

The same applies to the noise blanker: very low observable additional load on the processor in the Teensy SDR and also in the OVI40/mcHF.

db4ple, 2017-11-26:
@DD4WH: Due to the design of the load indicator you will not see the load caused by the NR code since this is NOT running in the audio driver. The load indicator has always only shown the audio_driver load as percentage of the full load (which is always 100%). It does not measure the "real" load of the UHSDR TRX. Technically, due to the software design, we have always 100% load, we are never idle.

db4ple, 2017-11-26:
My previous statement does not imply that the performance is not high, it just says, you can't prove it the way you did using the load indicator. See the FreeDV example: Here the load shown drops very much because almost all work is done outside the audio driver.  And believe me, FreeDV eats processor cycles for breakfast, lunch AND dinner.

DD4WH, 2017-11-26:
OK, thanks, you told me that before about the load indicator, but I forgot about that!

But the Teensy load shows me the real time used for these purposes (I know it, because I programmed it :-)), so I suspect it is not a problem of too much time used for noise blanker/noise reduction, because the Teensy load increases by only a few percent . . .

DD4WH, 2017-11-26:
Did a quick measurement in the Teensy SDR (sample rate 96ksps, 180MHz CPU speed):
CPU load increase:
1.4% for the noise blanker
6% for the noise reduction (I do not have the newest version on the Teensy, but a very similar version)
Both running in decimation-by-8, that means at 12ksps, exactly as in the UHSDR, where we have 48ksps and do a decimation-by-4 --> 12ksps for NB and NR.

DG9BFC, 2017-11-27:
just installed version 2.7.18 ... same thing ... new noiseblanker and/or new noisereduction producec garbage when switching on/off menue ... to have clear audio soemtimes i have to toggle menue on/off several times ... sometimes it needs 3 or 5 times ... sometimes even more ... sounds like the samples in the audio buffer get mixed up before they are sent to speaker ... for now i can NOT use that two functions in the mchf ... new nr and new nb do not work reliably (i just do not use them till we found that problem)

DG9BFC, 2017-11-27:
addition: when the sound is ok then the red led does flicker (like it should do when there is voice) ... when sound is garbled the led does light full on ... maybe that helps??

DG9BFC, 2017-11-28:
same problem on 2.7.19 ... spectral noise reduction is not usable ... when it works ... it sounds fine .... but it does not always work ... (most of the time it does not) ...
not only menue on/off can bring it back to live ... but also pressing prev/next in menue ....

db4ple, 2017-11-28:
Hi Sigi, 
you can wait for news on this issue. Until then, no need to do further tests.  We can reproduce the issue and are working on it. It may take a little time.

DG9BFC, 2017-11-28:
ok on that ... would you please so kind and explain a bit what does happen??? i am not a programmer but would like to know what is the problem .... what really puzzles me is that when it works ,,, i can also engage old nr ... when it does not work the sound is garbled (like on an old bad digital phone connection or so) ....

df8oe, 2017-11-28:
On 192KB RAM machines buffers used for NR are accidently overwritten by stack or other RAM using functions due to a lack of RAM. We can reduce possibly RAM usage on other places so that it may work in future on 192KB machines, too

DG9BFC, 2017-11-28:
ok ... stepped back to 2.7.15 (as that should have lower ram usage) ... works a tiny bit better (but also not reliable) ... when it works it sounds good ... but only WITHOUT snr gain smooth engaged ... if i switch on smoothing there is some noise added (sound a bit like hum noise and a bit like "digital noise" like our well known helicopter noise) ...
so ... when we have that ram usage solved i am guessing the smoothing should not be used (or i have to do further tests when we have it full working)

DG9BFC, 2017-11-28:
hmm ... now testing the  2.7.19 version again (with that long tone filter added) ... works similar good (or bad) as the 15 version (sometimes works and sometimes not) ... ok i think i just have to sit back and wait

db4ple, 2017-11-28:
This is what I suggested this morning already ;-) There are, for sure, more undiscovered issues, and your help is welcome. But for the NR thing, we are almost done with a probably sufficient fix but have nothing released yet.

DG9BFC, 2017-11-29:
GOTCHA! ... the version 2.7.22 produces fine audio and seems to be stable
toggling menue on off does nothing anymore with the audio
seems as you found the memory leak