df8oe/UHSDR #1219: Audio Muting again   BUG

WM0I, 2017-11-25:
THe audio muting has come back. Now if you push the PTT and let up on it the audio mutes and you have to turn the radio completely off and the Back on to get the audio to come back. This bug was gone for a long time but now it's back. really messes up a QSO or making first contact. Also the problem with the audio blasting when you change bands or change antennas through a coax switch you get the audio blast as if you had turned the audio up to max. All of this is back again.

db4ple, 2017-11-25:
Hi Richard,
please state the firmware version the issue has come back in, and for reference, the last one not having the problem. Of course only if you remember. Annd if the one with the issues is not the most recent one, please try that as well (OK, you answered that while I was typing, thank you for that!). 
Although there has been tremendous and quite impressive work by DD4WH and DL2FW on the new Spectral NR, this should not have an influence unless actually turned on, otherwise the code is kind off completely disabled and cannot harm. And apart from that have the recent release not seen audio related changes. Most of the work was display work.

73
Danilo

WM0I, 2017-11-26:
Actually it started with version 2.7.0, I didn't say anything then because of all the new updates to the firmware coming out.  Now I have too. 

> DD4WH and DL2FW on the new Spectral NR, this should not have an influence unless actually turned on, otherwise the code is kind off completely disabled and cannot harm. > 

How do I make sure the spectral NR is off?  What menu items control this?

Thanks in advance.
WM0I

WM0I, 2017-11-26:
Also I amusing the NR+notc  Set to 11.

df8oe, 2017-11-26:
Never could confirm and cannot actually confirm... :/

WM0I, 2017-11-26:
Not wanting to send you this unit to evaluate the problem I think I will make a few videos and send those to you. Would this help? What should I video other than going through the actions that the problem show up? 

WM0I

DD4WH, 2017-11-26:
Hi Richard,
is the problem still there if you switch off the old noise reduction and automatic notch filter you are using?
It is known that the NR & autonotch filter (the old versions) often cause problems.
73 Frank DD4WH

WM0I, 2017-11-26:
IWhat is replacing the old NR+Notch     =11 ?

DD4WH, 2017-11-26:
https://github.com/df8oe/UHSDR/wiki/Noise-reduction

We are working on the NR to make it work for the mcHF with small RAM and small processor. At the moment it could possibly work for you, please try it out. If it does not work, there is no other possibility than to wait for us to complete implementation for the small processors. BUT: it is still not sure whether it will be possible in the end.

73 Frank DD4WH

WM0I, 2017-11-26:
Sorry, the NR experiment did't help. I still have the audio muting and or blasting when changing bands.

For a short term fix the NR+Notch could be turned off when you press either the band+/- buttons then the band can change without loosing audio and then when the band has changed you can turn the NR+Notch band on.

Hope you can do this.

DD4WH, 2017-11-26:
we seem to have a communication problem.
Please do NOT use the NR + notch and switch it OFF.
When its OFF, is the problem gone?

WM0I, 2017-11-26:
Yes when the NR+Notch is OFF the audio muting is gone and the audio blasting max volume is gone when you change bands. Then after the new band is loaded I can turn the NR+Notch function back on.

THe quality of the audio is great with NR+Notch on.  I hope this function gets fixed. Sorry about the confusion. 

REgards Richard WM0I

df8oe, 2017-11-27:
Hi Richard,

old DSP NR and NB are performing very poor, are adding very big distortions to audio signal, do not remove any noise or peaks and are making mcHF instable (as you demonstratied). This was from the beginning of the project and has caused many, many issue reports.

We will remove this old functions as quick as possible and add some new, much better functions, which do what their name suggests: reducing and blanking noise without any impacts. These functions are already present in alpha stage in "expert/debug" menu, so please stay tuned if you do not want to participate in testing the functions. And: please do not use old functions "DSP NR" and "DSP NB". They are instable.

73
Andreas

closed 2017-11-27 by df8oe