df8oe/UHSDR #1222: Line in the grid

ly2bok, 2017-11-26:
There is no upper line on the screen in the grid.

df8oe, 2017-11-26:
I cannot confirm due to insufficient informations.

My test conditions:
Waterfall or scope mode /normal - not big).
Firmware version 2.7.16

Please be more exactly...

ly2bok, 2017-11-26:
http://www.qrz.lt/ly2bok/Laikinas/Grafic%20bug.jpg
So it will be clear.
Firmware version 2.7.16
Scope - big

df8oe, 2017-11-26:
:)
I do not know how it was designed before you noticed "the bug". I never use mode BIG because important ruler where I can touch magnify mode is missing if BIG is selected :)

So for me all is ok and I think we can live with the given layout - isn't it? It's only optical and does not impact functionality...

sp9bsl, 2017-11-26:
I think we lost it at the unification stage, it was graded colour line. Will add it later.

db4ple, 2017-11-26:
Hi,
yes, I commented the line drawing out. If we later want it back, no problem.

closed 2017-11-27 by df8oe