df8oe/UHSDR #1225: How do I make contact with DD4WH to give test data?

WM0I, 2017-11-27:
I have my test results for the new Spectral NR function.

 

1. First I turned the DSP NR+Notch off.
2. Then I went in the menu Expert/Debug section.
3. Turned  **Spectral NR** ON.
                                              Result:   The audio was choppy and the RED light came on steady.
4. Following the instructions from DDW4WH's web page I turned I adjusted the NR thresh higher
   higher trying to get the red light to go dark. Adjusted to the MAX value with no results of the red light 
   of going out. I let it set without doing anything else for about 10 minutes,  Still no results.

At this point I turned all the new items off. 
What to do and test next?
Best regards
WM0I

DD4WH, 2017-11-27:
Hi!
You seem to have done everything in the right way, thank you for testing!
However, at the moment, there is nothing you or we can do about that choppy audio. You seem to have an mcHF with small RAM (192kb), is that right?
If so, you have to be patient: the problem is known to us and we are working on it. Whether it will ever be solved, however, remains to be shown.
Sorry, but cannot give you a more satisfying answer . . . time will tell.
All the best and have fun with the UHSDR software,
Frank DD4WH

WM0I, 2017-11-27:
CPU    413h
ram    192kb

DD4WH, 2017-11-27:
Thanks! As expected. For these small RAM CPUs it is still not possible to use the new spectral noise reduction reliably. We are working on it and will keep you informed about progress.

WM0I, 2017-11-27:
Is there anything I turn off to give the cpu and ram more room to test again?

DD4WH, 2017-11-27:
no, I am sorry . . .

WM0I, 2017-11-27:
OK, I will wait for your team to make it work.

Thanks

DG9BFC, 2017-11-29:
hello wm0i ... install the 2.7.22 version ... and if it works you can close this issue (here that 22 version is working fine ... snr works good... seems to be stable

WM0I, 2017-11-29:
Whatever you did in version 2.7.22 works with in menu mode but when you come out of menu mode all the settings seem to go away and no noise reduction at all. Strange?

WM0I

closed 2017-11-29 by WM0I