df8oe/UHSDR #1238: 2.7.26: LCD issue

df8oe, 2017-11-30:
OVI40, 3.5" LCD: when switching to menu lower waterfall is not cleared. Cosmetical.

df8oe, 2017-11-30:
In 2.7.28 problem is much bigger. Now parts of S-meter box is erased and not correctly redrawn after leaving menu.

sp9bsl, 2017-11-30:
I see it too

db4ple, 2017-11-30:
Ok, my fault. The I mixed the coordinates. Sorry. One more.

df8oe, 2017-11-30:
fixed now.

closed 2017-11-30 by df8oe

sp9bsl, 2017-11-30:
Hi Danilo,
now better but frames around f buttons are disapearing sometimes after menu exit

db4ple, 2017-11-30:
Hm. Can't reproduce with 320x240 @ OVI40, but I have currently no 480x320 display / no mcHF for test. But I use the "official" screen area of the waterfall.

sp9bsl, 2017-11-30:
it happens always after the second menu display. Voltage indicator disapears also for a while. I'll try to debug it.

sp9bsl, 2017-11-30:
Have something, for the secon call of UiSpectrum_Clear the sd.wfall_disp_lines changes from 70 to 140

db4ple, 2017-11-30:
Hm. I think I know the problem then.

db4ple, 2017-11-30:
Well this can only happen it ts.spectrum_size is neither SPECTRUM_NORMAL (0) nor SPECTRUM_BIG (1).
Please check that. But how this is changed?

db4ple, 2017-11-30:
I have found the problem. It is the sequence of events. I will do a commit tomorrow morning to fix the problem.
In a nutshell: we calculate wfall_disp_lines before we actually calculate the correct size. And with that, we get the right result in the first call and wrong (doubled) results in all subsequent calls.

sp9bsl, 2017-11-30:
Ok, I always use SPECTRUM_BIG