df8oe/UHSDR #1241: Waterfall Display Issue

KF5PAL, 2017-11-30:
Noticed this with 2.7.28 (Remains in 2.7.29) Bootloader 3.4.4 on MCHF

Boot MCHF and waterfall displays fine.  Enter menu or change to Scope.   Try to return to waterfall display and it is just black with the filter BW indicator.  Only way to get waterfall display back is to reboot.

df8oe, 2017-12-01:
Cannot reproduce on 2.27.29.

- scope size normal
- scope light
- no magnify

Trying wirh touchscreen (my normal operation way) and "Touchscreen via menu".

KF5PAL, 2017-12-01:
Button use only.  MCHF turns on already set to Display waterfall - works fine
Press F1 for Menu softkey and menu displays 
Press F1 for Exit softkey and returns to main display but, waterfall ford not return, just a black background with my bandwidth display. (30% white and yellow line) scope still works fine

* scope size Big
* scope light- off 
* Magnify 2x

Antenna NOT connected during test

df8oe, 2017-12-01:
I have tried with exactly these settings on mcHF - cannot confirm. Waterfall appears when I leave menu.
So we must see if anybody else reports that issue.

KF5PAL, 2017-12-01:
In 2.7.30, the issue does not happen.  I also went back to 2.5.134 to compare the display of the waterfall. In 2.5.134 the waterfall display extends very far down (looks good).  In the later versions, including 2.7.30 the waterfall display does not go all the way down.  The bottom 2-3mm of the display are just black. (Scope looks fine and goes all the way down).

db4ple, 2017-12-01:
Hi,
yes, we are working a little on the display routines in order to support dynamic resolution switching and some of the small issues we will fix later once the main part (the ability to switch) is working. The waterfall size big@320x240 shrink is a "byproduct" of these modifications and once found and reported, can be fixed easily.

SergiuEpure, 2017-12-01:
IN 2.7.29 i have isue with waterfall, flicked all display. Actually the term waterfall is not corect because is going up, is actually like a fountain with water going up. If can be reverse the "waterfall" will be great. Thx

db4ple, 2017-12-02:
If all is fixed, please close the issue.

KF5PAL, 2017-12-02:
All fixed.

closed 2017-12-02 by KF5PAL