df8oe/UHSDR #1249: Working with these instructions doesn't seem to work for me.

WM0I, 2017-12-02:
https://github.com/df8oe/UHSDR/wiki/Noise-reduction

These instructions have me at my wits end. I follow them in order as in the instructions but I am not getting the results I thought I would get. I am still getting a very high noise level. I understand you can't get all the noise out but I thought I would get at least more than 10 to 15 % noise reduction.

I think you are doing a great first attempt as this problem. If I am not understanding as how to set up the New NR please point me to the correct way of setting up.

WM0I

df8oe, 2017-12-02:
It is under heavy movement. At the recent version 2.7.33/34 you only need to push M1.

Sorry - there will be no description until working on this part is finished. It is "try and error" - and you can get help here or (much more readers) at https://www.amateurfunk-sulingen.de/forum/

73
Andreas

DG9BFC, 2017-12-02:
ok then i give you some numbers
nr alpha 940-970 (stength of noise reduction .... 970 is already a strong setting)
long tone thresh 6000 long tone alpha 99900 (leave as is)
nr thresh 3000 (with vad "sohn" selected)
vad delay 10 nr beta 850 (leave as is)
.... try this numbers and report back .... (welcome back from dortmund andreas ... )

WM0I, 2017-12-02:
After making the settings change to what you instructed to do I closed the menu. I rebooted the unit and opened the menu to see if the settings stayed to what I set them too. They went back to settings I had before making the settings change. 

I made the changes again but the settings this time after the reboot.

df8oe, 2017-12-02:
The settings are experimental so they are not stored at the moment.

WM0I, 2017-12-02:
Version 2.7.35 is not allowing the auto notch filtering to work. Going back to version 2,7,30

WM0I, 2017-12-02:
For me version 2.7.30 has worked the best. Lower noise and notch filter is working also.
What I am going to keep a copy of the 2.7.30. I will download all the new daily's for testing and report to you as to the results of what I find.

WM0I

closed 2017-12-03 by WM0I