df8oe/UHSDR #1250: Porting to new STM32H743ZI

PY1CX, 2017-12-02:
Hi Everyone,

I'm thinking about porting the UHSDR to the new STM32H743ZI, which is the fastest ST Cortex M7.

Is there any porting guide so I can have a place to start?

Regards,
Felipe

sp9bsl, 2017-12-02:
Hi,
Please look [here](https://www.amateurfunk-sulingen.de/projects/ovi40-sdr#start). It is already done.
73 Slawek

PY1CX, 2017-12-02:
I found that OVI-40 isn't a complete project yet. How can I help?

db4ple, 2017-12-02:
Hi Felipe,
as Slawek said, we have UHSDR on F4 and since several month on F7. The next step is to go to H7 once it becomes mass available. 

Unfortunately there is no porting guide and the knowledge how to do a port quickly and in the right way is at the moment only with me, since I did the port to HAL in the first place and then to F7.

Compared to porting the firmware from F4 to F7, the port to H7 will be hopefully a lot simpler, since it is in many aspects almost identical to the F7 when it comes to peripherals (F4->F7 required a number of changes to due peripheral incompatibilities).

What would be your plans regarding the port? Do you have a H7 (board)? What is your experience level with STM32 processor on the lowest level and especially HAL and CubeMX? 

Our current plan is to port it to the OVI40 hardware (not very surprising). This means, we (for the moment) are only concerned with the "software" side, since we can use the same hardware definition.
Takes one complication out of the equation. 

How can you help? That is a good question! UHSDR is never going to be a completed project in the traditional sense I hope. And there are a lot of things to be done. With OVI40 F7 hardware ready (boards and kits will be available in the next days for preorder), the software side remains interesting. The new OVI40 hardware has SD card slots, we still need to implement USB configuration dumping / loading  and much more stuff. So if you have the right skills to do or to learn how to do it, you are welcome to our team. Many found a specific area of interest in the UHSDR software development.
It is an interesting and challenging task, a lot of fun and also a lot of learning.
If you look around and discuss with us, you'll soon have a topic of interest, I am sure.

73
Danilo

PY1CX, 2017-12-02:
Hi Danilo, Thanks for your anwser.

I have experience on STM32F0,F1 and F4 professionally, I have started with them with IAR, and since around 2015 I'm working with CubeMX. For 1 year I'm working with it alongside the Atollic IDE (HAL libraries).

I have a STM32F743ZI board with me, the Nucleo-144 one, I bought it on Mouser last week.  

I can help with firmware, hardware designing (PCB design on Altium). My goal with the H7 is that with all the power, it will be possible to do digital modes directly from the SDR, RTTY, FSK-31, etc.

Is there a list of the TODO of the project?

73 PY1CX - Felipe

db4ple, 2017-12-02:
Hi Felipe,

you shouldn't have trouble then. But F7 and H7 are a kind of different beast compared to F0,1,4 due to extensive caching you need to worry much more about things like order of events when it comes to communicate with external hardware. Hope the H7 does not have more tricks than the F7. 

TODO List: A lot of it is in the heads but a lot of it is in the open GitHub issues. Have a look.
I think, as the team grows, we have to think how to spread the knowledge who is doing what better.

RTTY -> works (just connect USB keyboard and set digital mode to RTTY)
BPSK-31 -> is being worked on (TX works, see #1002). Help maybe of interest. 
All about the supported digital modes is in the wiki and fairly accurate.
It also pays off to browse through the commit logs, we have a tradition of writing rather lengthy commit messages as means of telling others what goes on.

73
Danilo

DG9BFC, 2017-12-02:
would be nice if next step for digital modes would be bpsk31 receive

PY1CX, 2017-12-03:
Danilo, I'm studying the commits and how the code works. I just completed the first build on my own enviroment. In Linux Mint I had to add the gcc-arm-embedded PPA to my system to get the build working OK. It didn't worked for me on GCC 4.9. I had to upgrade.

I did the following link instructions to get the build enviroment working here: https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa

If you want to add it to the Wiki, maybe it will help other people too.

I'm also starting to develop a test bech to start coding here. As I doesn't have a mcHF yet, I bought a LCD and will start the development around it until we have a H7 PCB.

sp9bsl, 2017-12-04:
Hi Felipe,
H7 is the future. Now ST only promises the availability of parts, but not selling them yet, except one type populated on expensive evaluation board (my local ST office claimed delivery of engineering samples 2 months ago but without positive result as yet). Now we focus on F7 core which has a lot of horse power. If you want to help in development it would be good to have at least core+lcd(same like we use)+codec, so simply the UI board. The best one will be soon availabe (look at Andreas's pages for OVI40). For real work the RF board is also recommended (with working receiver at least).

73 Slawek

PY1CX, 2017-12-04:
Hi Slawek, The price of the IC is almost the same as the Nucleo-144 board.  I think the bare-IC is only $1 less than the evaluation board. In Digikey/Mouser they are rolling around $21 bucks and the Evaluation board is like $22.

My only problem finding any component now is around the LCD. I couldn't find a HY28 locally and if I  buy from China I will need to wait like 3 months until they arrive here. The codec I will get in my next Digikey order.

I will work locally with the H7 and after if you guys want my commits to roll over the main line, will be a pleasure.

sp9bsl, 2017-12-04:
Felipe, could you give a link where you bought it? I see on Digikey, Mouser and Farnell "0 on stock" for nucleo and single chips. 
HY28 is a lcd for Arduino, look for shops with parts for it.

db4ple, 2017-12-04:
Hi,
don't buy a HY28 (these are of bad quality these days), buy at least a HY32D or the RPi 3.5 ILI9486 Display 480x320. The RPi displays are readily available in Europe for shipping. Being SPI they require  only a few lines to connect. Be sure to get the ILI9486 variety. 

But definitely best would be to order a OVI40 UI and populate it with the H7 taken from the Nucleo board.
This is where my OVI40 H7 comes from (I donated my H7 Nucelo, we took it off the board and soldered it into the new one). 
A 480x320 display can be ordered with the board. 

Just for the record: Please do not be disappointed, if you start the port on your own, and then we may decide your code will not get into the main line. 
As I said, I will soon do the initial port to have all in the right structure. I am pretty sure, there will be issues with the port, but I am also pretty sure, most of these will turn out during operation on the real hardware only. Getting to boot, will be not trivial but with a working F7, I don't expect big obstacles here. 
Most of the work for porting is to write the correct linker script, setup the makefile to support another architecture and so forth. I did all of that already 2 times and I will be faster on that (prove me wrong :-) ). Once the initial port comes to available, the real work starts and any help is appreciate.
For the port itself, I will setup a separate branch on my GitHub and you are invited to work with me on it. 
I'll let you know then. Slawek did the same for his good work on the ILI9486 support, and it worked out great (@sp9bsl: Please, complain if this is not the truth).

73
Danilo

sp9bsl, 2017-12-04:
No, that's true, nothing to complain :)

PY1CX, 2017-12-04:
Slawek, I and a friend "pre-ordered" it based here: https://br.mouser.com/ProductDetail/STMicroelectronics/NUCLEO-H743ZI/?qs=sGAEpiMZZMve4%2fbfQkoj%252bOvguDsfUAxj950%252b8AGcc8I%3d

It's a lot of hype going on with the H743ZI and they are shipping when they arrive.

Danilo,
I got it! The common LCD around this same size here is the ones that uses the ILI9341 controller but I think it's easier to buy the OVI40 bare board with the LCD. 

No problem, I really agree that you will be faster with the porting. I will follow you with the porting, because I think that will be interesting.

sp9bsl, 2017-12-04:
Felipe, thanks. It is time to make a call to ST to remind them that I'm still waiting for the samples...

sp9bsl, 2017-12-04:
One more thing regarding LCD: do not buy something like [this](https://www.aliexpress.com/item/High-end-3-5-LCD-Touch-Screen-Display-Module-Board-320x480-RGB-For-Raspberry-Pi-2/32611946526.html?spm=2114.search0104.3.174.6JgzE5&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_5000015_10151_10344_10068_10345_10342_10343_10340_10341_5130015_10541_10084_10083_10304_10307_10301_10539_5080015_10312_10059_10313_10314_10534_100031_10604_10603_10103_10605_10594_5060015_10596_10142_10107,searchweb201603_14,ppcSwitch_5&algo_expid=8b0015b8-d69a-43a5-b3a7-f083b5870b41-22&algo_pvid=8b0015b8-d69a-43a5-b3a7-f083b5870b41&rmStoreLevelAB=4)!
It doesn't have the rs pin which is mandatory.
[This](https://www.ebay.com/itm/3-5-inch-B-B-LCD-Touch-Screen-Display-Module-320-x-480-for-Raspberry-Pi-V3-0/292287703817?_trkparms=ao%3D1%26asc%3D41375%26meid%3Dd24e3759ffae4b7eab4ebf5dfb95513c%26pid%3D100705%26rk%3D1%26rkt%3D1%26&_trksid=p2045573.c100705.m4780) one is working ok.

73 Slawek

PY1CX, 2017-12-04:
The second one I can easily find here in Brazil! Will get one of this!

db4ple, 2017-12-05:
Hi,
a) preorder is open for OVI40 board and LCD, be quick to get one from the first batch. Go to:
https://www.amateurfunk-sulingen.de/forum/index.php?board=18 there you'll find the forum and the details.

b)  My OVI40 H7 does turn on the red led, unfortunately it does not recognize the display and after that it gets worse. But good news I am able to compile, flash and debug.
See db4ple/mchf-github, ovi40h7 branch. As I said, not usable right now but we should get it to work from there. Full build of current firmware is possible without problems in Eclipse. 
Did not care yet to setup the make build, but this is also not so difficult.

df8oe, 2017-12-06:
Congrats to H7 porting work. I think there are some small but important different settings for CubeMX which are missing yet. But H7 will not be available within short time span so I think porting has "normal priority" and will be finished in time.

PY1CX, 2017-12-06:
Hi Danilo,
I will get a PCB out of the first batch as soon as I can. I'm now travelling for work and can't get it.

DF8OE, 
It seems that H7 is rolling out in January.

73 Felipe PY1CX

df8oe, 2017-12-06:
Hi Felipe,

I have reserved an UI PCB for you - but pse do make an order on my webpage because of you will get an order no. That is important for my processing ;)

73
Andreas

PY1CX, 2017-12-08:
Andreas,
I did ordered yesterday. Thanks

db4ple, 2017-12-08:
Hi,
I have (kind of bad) news for you: The initial port to H7 is almost done. I have to verify quite a number of things but it seems that all peripherals are basically working and should, coupled with a RF board, be able to receive and transmit. Thanks to the extremely similar peripherals and our more or less consequent use of CubeMX and HAL the port required surprisingly small changes to the existing code once the CubeMX configuration is made correctly.  There is still enough work to be done to unleash the full power of the H7 but we have starting point.

closed 2018-01-30 by df8oe