df8oe/UHSDR #1252: Bricked radio after config backup

g7jiq, 2017-12-02:
Has anyone had this problem and how do I rectify it.
Having updated to the latest software, (Have been following them for some time)
I decided to backup the config data for the first time, having read that ths should be done.
Having selected backup config, I seletted DO IT!! 
The radio emitted sevral desending tones and returned to the menu.
All seemed ok until I power cycled it,
All I get now is a white screen if I hold the power button, No other buttons make any difference.
I tried the usb update button combination an all I get is a white screen which goes off when the power buttons released.
Also tried reprogramming the boot files, but the usb is not responding/working.
Has anyone got any ideas, ?
Has this killed the cpu?

db4ple, 2017-12-02:
Hi,
good news is; With config backup it is impossible to kill the CPU. 
Now: I assume you don't have a I2C EEPROM fitted, right?

db4ple, 2017-12-02:
Next question: Which bootloader do you have?

g7jiq, 2017-12-02:
Yes I have, and I have V3.4.4 boot loader

db4ple, 2017-12-02:
Ok, you don't have an EEPROM and your bootloader does not respond to Band- + Power or Band+ or Power.
Is that correct?

g7jiq, 2017-12-02:
Just looked and I have a 24LC1026 fitted and did show in the menu as recognized.

g7jiq, 2017-12-02:
No response to the power + Band+or-

db4ple, 2017-12-02:
If this is the case, I am afraid you have to go through the initial bootloader install procedure:
Yes, I now checked. We do the right thing and permit backup only of there is an EEPROM. Makes no sense otherwise. 

I am afraid, you have to run the procedure linked below for initial bootloader install. But do not despair, the configuration is still in the eeprom, so even installing all from scratch should get you back where you were (we have to look into the Backup issue once we got your TRX working again):

https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/useful_manuals/bootloader_install.pdf

Since you have the TRX open anyway, should not take long. If it does not respond to the P6 procedure it is more serious as this indicates indeed a damage of the MCU but not from the backup config itself. Do you have stable power connection? This can be fatal, if you have power fluctuations during flash writes (Backup config is nothing else).

Try and report.

g7jiq, 2017-12-02:
Ok thanks for the info, will try this and report back.

g7jiq, 2017-12-02:
Have tried to reload the boot loader as suggested, but the pc (tried 2) does not see the usb pugged in,
Looks like the MCU is dead....
(Main psu is a nice stable 12v)

g7jiq, 2017-12-02:
An update, MCU programming link (P6 jumper) wasn't making contact properly, Having tried another link the USB worked and was seen by the DfuSe program,
I have now re-flashed V3.4.4 and the radio is now working.

Thankyou very much for your help..

db4ple, 2017-12-02:
Can you please try once again the configuration backup just to see if the issue can reproduced?
Thank you!

g7jiq, 2017-12-02:
Just tried a config backup and restore (twice) and it worked fine, and was very quick compaired to the 30 seconds it took when it failed.
Strange one..

db4ple, 2017-12-02:
Ok. Then we take this as a glitch and you may close this issue now. Thanks for trying again!

g7jiq, 2017-12-02:
Must have been, 
Thanks again for your help.

closed 2017-12-02 by g7jiq