df8oe/UHSDR #1257: STM32F4xx clock frequency - question

s53dz, 2017-12-03:
I have promoted the mcHF STM32 from F407 to F429. Mainly to get 1 MB flash and 256 kB of RAM. But it also means that I have got a 180 MHz core now. I wonder if it is possible to count this fact in to get a small speed improvement (180 / 168 = 7 %)?
This is a table showing the different usage of internal PLL for that and consequently changing peripheral buses:
https://stm32f4-discovery.net/2015/01/properly-set-clock-speed-stm32f4xx-devices/

73 Bojan

db4ple, 2017-12-03:
Hi,
I can answer with this one:

https://stm32f4-discovery.net/2014/08/library-31-usb-hid-host-for-stm32f4xx-devices/

Read to the part where they say, 168Mhz since with 180Mhz USB is not possible.

And with 7% you will not gain much speed, you may even loose some if we have to access flash slower (Didn't check if this is required for 180 Mhz in the datasheet)

BTW, it is a hell more complex to get all the clocks right (for instance we have also I2S at 12,88 Mhz which must be possible).  Even with CubeMX clock tool, it is not too much fun to get all the stuff right.

Get yourself an OVI40 UI :-) it runs at 216 Mhz and is twice as fast at the mcHF but then you need a new front (at least). Anyway, don't think we should go this route unless some STM32F4 wizard tells us the right clock configuration. BTW, you can even overclock these but again someone needs to find out if there is any real gain to be had. Even 20% faster will not bring too much. And we have to optimize for 168 Mhz anyway due to the large number of mcHF users stuck with 168 Mhz.

s53dz, 2017-12-04:
Danilo,

Thanks for the explanation. So, it is a lot of delicate work for maybe some % of performance. And to be noticeable we would need more. About 180 MHz and USB, I am not sure. PLL can do it. But, maybe I am wrong.

Excerpt from your link:
> You STM can go up to 180MHz.
> Set PLL_M to 26 and HSE_VALUE to 26000000. Then you will have 180MHz and USB will also work correct.
> 

I am following the OVI40 development so no need for advertisement here. 

With this I close my question.

TNX 73 Bojan

closed 2017-12-04 by s53dz

db4ple, 2017-12-04:
Hm, interesting. I have to see if at least 180 and USB is possible with 16Mhz. The guy should have changed the text in the document if he says in the comments, 180 and USB is possible.
Internet wisdom is always right, isn't it. Nevertheless the gain would be minimal.