df8oe/UHSDR #1259: Version 2.7.36 looks like a winner.  Report information.

WM0I, 2017-12-03:
I loaded the new version 2.7.36 in and I only had to adjust the volume to a comfortable level.
Also most of the noise has changed to a different tone ( more like soft white noise ).
I will keep test most of the day and then report again.

WM0I

WM0I, 2017-12-03:
In further test I found both 2.7.36 and 2.7.37 to have lost the notch ability no matter the settings are.
Went back to 2.7.30. If you could combine the notch filter function of 2.7.30 with the noise reduction of  2.7.36/37 then you will have a great NR.  I will still be downloading all the new dailies for testing.

WM0I

WM0I, 2017-12-03:
Sorry, I forgot to mention I am using the ( DSP A-NOTCH ) to kill the carriers when using 2.7.30.
This ( DSP A-NOTCH ) filter is not working on the 2.7.36/37 versions. 

Best regards
WM0I

df8oe, 2017-12-03:
Yes, that's right. We are working on NR functions recently and other DSP functions like notch or peak may be temporarily unavailable during this development. They will return when DSP NR has stepped forward.

73
Andreas

DG9BFC, 2017-12-03:
there is a notch like filter added ... name is long tone filter .. not as good as the old notch filter but usable (and they are working on it so need reports of the function from some users) ...
last version that has old and new nr awailable (and so a working notch) is 2.7.31 ... but you should try the  36 37 (and any further) and report your findings (also on the notch and not only on the nr filter)
....
what i did find out is if you use nr plus lt filter and turn the lt filter thresh down ... there is a point where audio is nearly silence (also the hiss is gone) and only voice gets to speaker ... sounds a bit like fm voice (but with too much high tones) .... with a bit of finetuning of bass and treble it does not sound that bad (and you have an almost noisefree voice) ... try out the settings and report your findings .... only that way the programmes can make that function better

WM0I, 2017-12-03:
For some reason I was not able to adjust the Lone Tone to get the you said I could get on version 2.7.36/37.   Neither the noise was reduced nor was the notch working. I am not sure why I can not get the results you get. I will try again on the next version.

Best regards
WM0I

DG9BFC, 2017-12-03:
nobody said the new notchfilter works like the old notch filter we had before ... it is more like we have 2 different strength of noise reduction (with nr and nr+lt filter) .... these are switched on with nr or nr+notch on the front (and values are set in menue) .... yes it is not a real working notch but a noise surpression ....
tones are not completely filtered out like a real notch but damped somewhat ....
yes i also keep the 31 version .... but it is worth trying out the new noise surpression ...
it works also on the 31 version but can not be switched on and off from the front (only in menue)

WM0I, 2017-12-04:
On my side of things 2.7.30 is better than 2.7.31. I have no idea why. All my test point me to keep 2.7.30.

I will keep checking the new versions.

closed 2017-12-04 by WM0I