df8oe/UHSDR #1263: SSB TX filters please

lb5va, 2017-12-04:
Hello, LB5VA Stig here; I would like to get to chose between at least 3 deifineable SSB TX filters defined the same way as the RX filters. To choose them just press PTT and BW-button. Possible?

DG9BFC, 2017-12-11:
hmmm we already have 3 tx filters selectable in menue ... they can be changed easy so why you want them on the screen (selectable with a button while txing)???

closed 2018-01-31 by df8oe

lb5va, 2018-02-01:
Sorry to open this again. ..."we already have 3 tx filters selectable in menue"
You are talking about the SOPRANO/TENOR/BASS?  Is'n that more like equalizers.
I'm talking about configurable TX bandpass filter. The same way as the RX filter is configured.
Like most SDR-radios can do!  
But if no other wants this I rest my case....
- Stig

DD4WH, 2018-02-01:
Hi Stig,
well we are working hard on our capabilities, but we cannot change hardware filters in software ;-).
Please give an example of an SDR radio that can change the hardware TX bandpass filter via Software, I would be very surprised :-).
And in fact the SOPRANO/TENOR/BASS filters are audio bandpass filters . . .So what are you missing exactly?
https://github.com/df8oe/UHSDR/wiki/How-does-your-UHSDR-software-DSP-work
(look at the paragraph "TX-transmit path"):
"All TX IIR filters are Bandpass filters with ARMA structure IIR filters of 10th order elliptic with 60dB stopband response with precalculated coefficients [fdatools in MATLAB]. There are three TX audio filters the user can choose from:
SOPRANO: 250-2950Hz
TENOR: 150-2850Hz
BASS: 50-2750Hz"
AND you have an additional TX equalizer for TX bass & treble (which nobody seems to know about and use, it seems ;-))
So again, which (software!) feature are you missing? 

73 Frank DD4WH

lb5va, 2018-02-01:
Thanks Frank, for clearing this up for me, and for the link to information.  I'd never guess those three filters (with name) was hardware filters. I thougt the TX bandpass was all done in DSP/SDR software.
And - I did not know that the Bass and Treble worked for TX, to be adjusted while TX. Tried that first now. Thanks!  
Well, I'd like to have posibllity for TX bandwidth 50/100 - 3000/3500/4000 for essb experiments in local qso's.

DG9BFC, 2018-02-01:
those filters in the menue are soft filters ... the hardware filters are the lowpass filters in the pa stage
you can use the "bass" setting (i think you wanna have more OUMP on lower end?!?) ... plus turn up the bass setting on front while txing ... tx filter wider on upper end will not come i guess  ... cause it is not ham friendly to use more then 3 kc bandwith in ssb