df8oe/UHSDR #1264: Recent 918SSB / mcHF clone  no power off after firmware upload

mariomax232, 2017-12-04:
Hello, I have one Recent 918SSB / mcHF clone and I upload the last firmware 2.7.39 and bootloader 3.4.4, all runs ok but when I power off  2 second later the transceiver  power on again .
With old firmware and bootloader the transceiver power off perfect.
Old firm---2.5.130
Old Bootloader---3.4.0
I replace the U3 ic  LM2941 with a new LM2941CT and the problem continued.
Any idea to repair this problem??
Thanks......Mario....EA7IXD.....73!!!!!!

DD4WH, 2017-12-04:
sorry, no support for chinese clones here from my side . . .
73 Frank

mariomax232, 2017-12-04:
Hello, thanks you of all forms.

WY5Y, 2017-12-05:
Did you save the old firmware? You could reload that one.

df8oe, 2017-12-06:
This must be a hardware issue because "offcial mcHF" does not have this behaviour.

closed 2017-12-06 by df8oe