df8oe/UHSDR #1265: Noise when using dual screen or scope screen

ac0ax, 2017-12-04:


ac0ax, 2017-12-04:
I think I found it. When I turn RX NB and DSP NR to 0 the noise quits.

ac0ax, 2017-12-04:
IF I select the DSP fuction the noise returns. It also returns if I turn on the New noiseblanker.

lb5va, 2017-12-04:
Just confirming what ac0ax writes about the TX led. On push on PTT and it is off again.

Jmelz, 2017-12-05:
Same as above here.

Jmelz, 2017-12-05:
Ac0ax that is.

df8oe, 2017-12-06:
Cannot confirm at mcHF or OVI40...

DG9BFC, 2018-01-07:
popping noise may come from noiseblanker ... but the complete nr part is in heavy development so i am guessing you can close this issue here ... try out newer fw ... we are now at 2.7.60 so your 39er version is a bit old ... grin (sometimes there are 3 or 4 version numbers a day so check regularly and report the issues allways with newest fw (or with report what changed between newest and version before)

ac0ax, 2018-01-08:
I just installed the newest version and it seems to be gone now. thank you.

closed 2018-01-08 by ac0ax