df8oe/UHSDR #1266: Trying to do a complete the entire Adjustment and Configuration manual. Help

WM0I, 2017-12-05:
First:  Setting the PA Bias I get different results that can be between 31 to 81 depending which band you are making the PA Bias in. This causes the attempt of setting each band to 5 watts hard to do. If you can't back the power down to 5 watts you have to go back to the PA Bias and lower the value so you can set the power to 5 watts. This happens to every band and changing the PA bias level for one band makes the other bands wrong.

Is there a better method in setting the PA Bias that will satisfy all the bands without this problem. 
This problem also makes setting the TX IQ's correctly.

I need help with this please!

sp9bsl, 2017-12-05:
Hi Richard,
there is no need to have bias setting for every band. This is the first step of callibration at the very beginning of work with PA. This is simply the first rule for the mosfets! Then you set the apropriate level of signal for 5W and maximum power. If you start to change the bias instead of signal level, you enter in very non linear mode. I have bad feeling that there is something wrong with your PA if you need to do it for every band.

73 Slawek

WM0I, 2017-12-05:
Yes I now see a problem with the PA. I need a schematic of the PA board. Where do I find one?

WM0I

sp9bsl, 2017-12-05:
You have to check the pcb version number on RF board and visit M0NKA site, he has schematics for all available versions in download tab, please search there. If you have a clone - RF board is more or less similar to v0.4.

hazzer23, 2017-12-09:
I have simular problems setting bias current and theybeing unable to set the RF power across the bands, its a clone radio, I suspect the RF mosfets are the problem. I have ordered replacements form a known good source,

Tibet4690, 2017-12-10:
Have a clone myself, following this topic with interest! Been meaning to go through and verify my TX power settings, you have me worried about what I'll find when I get around to it.

closed 2017-12-24 by WM0I