df8oe/UHSDR #1267: Menu protection for destructive options

frazierjason, 2017-12-05:
There are a few menu options that take immediate and irreversible "Do It!" actions.  These should have some safety precautions against unwanted selection.  Not having yet read the sources, I'll guess the menu structure doesn't accommodate a confirmation dialog.  An easier protection scheme could be done, a la Android Developer Mode, where you have to keep triggering (scrolling right) for seven times to count down to the real "Do It!" option before it will be triggered.

I'm happy to put my rusty C skills to use and submit a patch, if someone can pitch the preferred design change or point me at the correct sources to get started on.

TL;DR
I received a preconfigured genuine mcHF from another ham a couple weeks ago, hooked it up, and within the first one minute of use, I had a noisy RIT encoder trigger while I was scrolling through the menu.  It just happened to unluckily trigger on "Standard Menu" >> "Restore Config (Do It!)", on a unit that had never seen a flash backup before and always used EEPROM.  This seems to have resulted in a loss of EEPROM config data in exchange for all defaults, and now I have to partially disassemble and recalibrate the settings myself with a couple borrowed analyzers.

Jason Frazier W7DM

df8oe, 2017-12-06:
I think this is a very good idea. We will do it. There is already confirming routine for loading defaults (press band+ and band- simultaneusly to confirm, any other button cancels operation). Maybe we will choose this as "standard confirmation" in future.

73
Andreas

frazierjason, 2018-02-11:
I would like to implement this.  I heard from another mcHF user on the Yahoo group that keeps accidentally erasing his settings.  Any comments or directives before I dig in?

df8oe, 2018-03-18:
what is the progress in this work?

frazierjason, 2018-03-23:
I haven't done anything for this one (yet).  I've got a lot of non-ham activities and also major spring cleaning to tame the junk in my garage.  It will probably be at least three weeks before I can come back to looking at the UHSDR codebase.

df8oe, 2018-06-26:
I have found a comfortable solution for it in commit #e31a29d6 (version 2.9.39). Encoder behaviour was fixed some weeks ago but I think securing dangerous actions is a good thing.

closed 2018-06-26 by df8oe