df8oe/UHSDR #1268: Dual-Display: reduce the heigh of Waterfall/Spectrum via touch on the fly

DL8EBD, 2017-12-05:
with Dual-Display we can see spectrum and waterfall simultaneously.
I would like to have an option to reduce/increase the high of waterfall or spectrum via touch and slide on the fly.
flatter waterfall = higher spectrum and vice versa

thx
Thomas, DL8EBD

sp9bsl, 2017-12-05:
It is a good idea to have both resizeable. User can in this way speed up scope with SPI mode for 3.5" lcd, decreasing the waterfall height.

73 Slawek

ly2bok, 2017-12-06:
Slawek, I also agree with such a variance. It would be very convenient.
Justas LY2BOK

DG9BFC, 2017-12-16:
if you then completely slide the border up or down you have just one view left (onlys spectrum or waterfall) ... then the switching between both views could be deleted completely ... you have then only one screen left (where you slide the border up or down depending if you wanna see spectrum, waterfall ... or both) ... nice idea :)

sp9bsl, 2017-12-16:
Yes, we'll see how this works (the refresh rate when sliding, especially for SPI lcds), if it will be ok, we remove it from menu and touch (less items=better ;) ). But be patient please, there is a lot of work with all other features that have higher priority :)

sp9bsl, 2018-01-26:
Hi,
Since 2.7.69 we have this working: #1310
So, this issue can be closed.

73 Slawek

ly2bok, 2018-01-26:
Thank you, Slawek. Everything works perfectly. A very good solution was found. Thank you.
Justas LY2BOK

sp9bsl, 2018-01-26:
Hi,
this is positive result of "mass thinking" :) Thank you all for that!

73 Slawek

closed 2018-01-30 by df8oe