df8oe/UHSDR #1269: S-Meter Calibration. ( Question from the Yahoo Group )

WM0I, 2017-12-06:
I found this question on M0NKA mchf Yahoo group.
I have the same problem. I did not find an answer to the question so I am Quoting the question.

> Can anyone direct me to a procedure how to adjust the S-Meter, please.
Without anything connected to the antenna jack my mcHF indicates S3,
when connected to a dummyload it indicates S5.
Thank you very much1
regards
DK3HE, Eberhard

> 
So I hope the gurus have a way to calibrate.

Best regards
WM0I

DD4WH, 2017-12-06:
Hi!
the UHSDR-Wiki is your friend, many topics are treated in our UHSDR Wiki:
https://github.com/df8oe/UHSDR/wiki/S-Meter-&-dBm-Hz-display
Calibration is not an easy task and you would need a calibrated signal generator to do this.
73 Frank

DK3HE, 2017-12-10:
Hi,
I have now read the WIKI "S-Meter-&-dBm-Hz-display", and will need some time to fully evaluate
it. Dear experts, please wait before spending your valuable time for my question. I will reconsider my question.
Thank you for the response anhow!
vy 73
Eberhard

closed 2017-12-14 by WM0I