df8oe/UHSDR #1273: 2.7.41 ?   = Noise Reduction ?

WM0I, 2017-12-10:
I loaded 2.7.41 and found a few things missing in the debug/expert menu. I tested the ability reduce the noise and notch filtering with results not being very good. I tried to get the noise reduction just by reducing the NR Threshold down slowly to the point of no or very little noise and just the voice allowed through. Could not get the results I got from 2.7.39. 

I went back to 2.7.30 and lowered the AGC down to 10 and selected the NR+NOTCH to 10 also and I got audio which sounds almost like FM radio. As the audio signal got stronger the overall damping of the noise was cumulative to the value of signal if high. If the signal started to become less the noise did not immediately become noisier it took a couple of minutes to for the noise to return.  

I was really hoping the new NR + NOTCH filter worked in version 2.7.41.  

In version 2.7.41 Not being in the menu's I got audio sputtering and the red transmit flashed in time with the sputtering audio. Tried to adjust it out with what left in the menu settings with no avail. the only time the sputtering stopped was when I entered the menu mode.

I hope this helps.
Best regards 
WM0I

df8oe, 2017-12-10:
Actually we are collecting informations. No changes were made in the last three versions regarding noise blanker. 

Problem is it seems to depend on many, many settings how it works. That makes improvements more complicate. I never have any sputtering - cannot confirm. Noise reduction on QSOs where signals are varying much is not an easy task. I can confirm that best results for weak signals differ from those for strong signals. Many thanks for testing - every information is useful. Sometimes it takes longer time to find a way. Longer duration does not mean there will be no more states ;)

73
Andreas

db4ple, 2017-12-10:
Hi WM0I,
to help you and others to understand what changed in newer releases, a good way is to check the commit messages here https://github.com/df8oe/UHSDR/commits/active-devel . We try to give sufficient in the commit messages, and you will also learn who commits what normally. For instance you will learn that I (DB4PLE) in general not mess around with anything related to audio signal processing. That would be for instance @DD4WH's or @dl2fw  commits you have to watch out for. But of course this is not a strict rule. So have fun reading our commit messages.