df8oe/UHSDR #1275: text output bootloader with ILI9486 LCD

df8oe, 2017-12-10:
I am not sure: is it possible to add dual LCD support to OVI40?

And if we work at this part:

Might be better to expand bootloader space for OVI40 to 64KB and remove virtual EEPROM support? We will have sdcard support in future and H7 will not offer possibility of virtual EEPROM at all - so why not unifying?

db4ple, 2017-12-10:
Of course, we can do everything. Right now the rule is: as long as it fits into 32k. Which it does. 
All bootloaders build with LCD_TYPE=2 should work. Travis builds with this configuration.
Bootloader is only 320x240 but this is a minor issue I would think.

df8oe, 2017-12-10:
Yes - so it is. I haven't verified this and use "old" bootloader. I will build new one and test.

Regarding other "wishes" this has very low priority and can be at horizon of 2do list ;)

closed 2017-12-10 by df8oe